platformio/platformio-core #614: update esptool

boneskull, 2016-04-12:
What kind of issue is this?

- [x] Bug report. If you’ve found a bug, please provide information below.

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS X El Capitan

**PlatformIO Version** (`platformio --version`): 2.8.6

### Description of problem

I have not had any success with the packaged `esptool` for uploading to a WeMos D1 Mini.  It *always* fails.  I replaced it with the master branch [here](https://github.com/igrr/esptool-ck) and now uploads work.

#### Steps to Reproduce

1.  Get a Mac with El Capitan, D1 Mini, and install driver from wemos.cc.  Reboot
2.  Build anything with platformio
3.  Try to upload

### Actual Results

```
$ pio run -t upload
[Mon Apr 11 17:11:26 2016] Processing d1_mini (platform: espressif, lib_install: 18, 19, 89, board: d1_mini, framework: arduino)
----------------------------------------------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs/d1_mini/firmware.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.wchusbserial14220
"/Users/boneskull/.platformio/packages/tool-esptool/esptool" -vv -cd nodemcu -cb 115200 -cp /dev/cu.wchusbserial14220 -cf .pioenvs/d1_mini/firmware.bin
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.wchusbserial14220
espcomm_upload_file
espcomm_upload_mem
opening port /dev/cu.wchusbserial14220 at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
resetting board
trying to connect
[... 255 lines cut ...]

    taskManager.StopTask(&taskKeepConnected);

    Serial.println("Connecting to " + String(SSID));
    WiFi.begin(SSID, PASS);

    taskManager.StartTask(&taskWaitForConnection);
  }
}

void setup() {
  Serial.begin(9600);
  WiFi.mode(WIFI_STA);
  dht.begin();
}

void loop() {
  taskManager.Loop();
}

```

### Additional info

I have had some success uploading the built `.bin` with `esptool.py`.

boneskull, 2016-04-12:
oops, wifi password.

valeros, 2016-04-12:
Hi @boneskull !
We use the latest esptool (0.4.8) from Arduino ESP8266 (2.1.0) [package](http://arduino.esp8266.com/versions/2.1.0/package_esp8266com_index.json).
Does uploading to your board work well in Arduino IDE? Which USB-UART bridge are you using?

MichMich, 2016-04-23:
I've got the same issue as well. Impossible to upload to the D1 Mini. The Arduino IDE manages to upload from time to time, but gives a lot of problems too ... Maybe a driver issue for the D1's CH340G USB-UART bridge?

ivankravets, 2016-04-23:
Maybe need to ask @wemos or http://forum.wemos.cc ?

MichMich, 2016-04-24:
There definitely is a difference between Platform and the Arduino IDE. The Arduino IDE manages to upload 25% of the time (so, it definitely has issues, but succeeds from time to time). Platform IO never manages to upload.

valeros, 2016-04-25:
Could you please try this [esptool](https://github.com/igrr/esptool-ck/releases/download/0.4.8/esptool-0.4.8-osx.tar.gz)? Just replace old one in `/home/.platformio/packages/tool-esptool`
Have you tried `esptool.py` as the alternative uploader?

MichMich, 2016-04-25:
The esptool you link to, appears to be the one I'm using. I tried using esptool.py:
````
esptool.py --port /dev/cu.wchusbserial1440 write_flash 0x00000 .pioenvs/d1_mini/firmware.bin
````

But unfortunately, it fails to connect:
````
Connecting...
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 5, in <module>
    pkg_resources.run_script('esptool==0.1.0', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 492, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 1357, in run_script
    exec_(script_code, namespace, namespace)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources.py", line 47, in exec_
    exec("""exec code in globs, locs""")
  File "<string>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 471, in <module>
    
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 149, in connect
    
Exception: Failed to connect
````

So this isn't a Platform issue, but really an issue with the WeMos. :(

MichMich, 2016-04-25:
It turned out I was using a very old esptool.py, after upgrading esptool.py I am now able to upload using platform.io.

MichMich, 2016-04-25:
Interesting. I found this pull request for the regular esptool: https://github.com/igrr/esptool-ck/pull/29
I tried that pull request, and that one seems to work fine. Hopefully they merge that pull request soon.

boneskull, 2016-06-16:
I can confirm the PR @MichMich mentions fixes my problem.

ivankravets, 2016-06-16:
@MichMich @boneskull , please contact with @igrr and ask him to merge this PR https://github.com/igrr/esptool-ck/pull/29

MichMich, 2016-06-16:
Done!

boneskull, 2016-06-21:
@ivankravets I believe the fix was merged in esptool

ivankravets, 2016-06-21:
@valeros the esptool has been released a few days ago. Please update all packages https://github.com/igrr/esptool-ck/releases/tag/0.4.9

valeros, 2016-06-21:
Updated, thanks!

closed 2016-06-21 by valeros