platformio/platformio-core #615: lptm4c1294ncpdt target (titiva/energia) can't find the SPI library

jasiek, 2016-04-13:
- [ ] Bug report. If you’ve found a bug, please provide information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Darwin jasiek.local 14.5.0 Darwin Kernel Version 14.5.0: Mon Jan 11 18:48:35 PST 2016; root:xnu-2782.50.2~1/RELEASE_X86_64 x86_64

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.8.6

### Description of problem

When trying to build a sample project using Energia, the build system fails to find the SPI library (perhaps it isn't included in your Energia bundle?)

#### Steps to Reproduce

1. Clone https://github.com/londonhackspace/acnode-cl/tree/platformio
2. Build the project, it succeeds.
3. Remove the vendored lib/SPI which I copied over from the energia/Energia repository.
4. The project fails to build with:
```
.pioenvs/lptm4c1294ncpdt/SD/utility/Sd2Card.cpp:29:1: error: 'SPIClass' does not name a type
SPIClass SPI_for_SD;
^
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:lptm4c1294ncpdt]
platform = titiva
framework = energia
board = lptm4c1294ncpdt
```

**Source file to reproduce issue:**
see `src/ACNodeClient.ino`

valeros, 2016-04-13:
Hi @jasiek !
Actually, it isn't PlatformIO issue, because one of the standard `Energia` libraries (`AIR430BoostEuropeETSI`) contains strange empty `SPI.h` file, so builder decided to use it. Please update framework package via `pio update`.
Thanks!

jasiek, 2016-04-13:
I couldn't work out why it needed to build that library as I don't have any explicit deps on it. Is there a dump somewhere that would indicate why it was scheduled for building?

jasiek, 2016-04-13:
Also, thanks for your help - very much appreciated.

valeros, 2016-04-13:
Probably in PlatformIO 3.0 such debug dump will be implemented.
Let us know if you need any help.

closed 2016-04-13 by valeros