platformio/platformio-core #619: Feature Request: support Visual Studio Code

dieseltravis, 2016-04-15:
- [X] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:
* Windows/OSX/linux

### Description of problem

VS Code is based on Atom but performs much better.

### Additional info

Here is more info on VS Code Extensions:
https://code.visualstudio.com/docs/extensions/overview

and publishing to the VS Code extension gallery: 
https://code.visualstudio.com/docs/tools/vscecli

marvinroger, 2016-06-25:
What do you mean by "performs much better"? Atom is an amazing IDE, IMO.

dieseltravis, 2016-06-25:
Atom is good, but can be slow. Code is very speedy.

marvinroger, 2016-11-15:
Coul you please "*like*" the first post instead of publishing *+1*s? Don't forget we receive an email every time something is posted. Thanks. 😉

zgoda, 2017-01-31:
There's a project for this on GH but it seems inactive https://github.com/fabianofranca/vscode-platformio

formulahendry, 2017-03-17:
Hi friends, I have just created a [PlatformIO extension](https://marketplace.visualstudio.com/items?itemName=formulahendry.platformio) for VS Code. You could install and have a try. 😃 
In the initial release, it supports below basic function:
* Build PlatformIO project specified in [Project Configuration File platformio.ini](http://docs.platformio.org/en/stable/projectconf.html#projectconf)
* Upload firmware to devices specified in [Project Configuration File platformio.ini](http://docs.platformio.org/en/stable/projectconf.html#projectconf)
* Open Serial Monitor

Currently, the extension is in the very initial phase. If you find any bug or have any suggestion/feature request, please join the chat on [Gitter](https://gitter.im/formulahendry/vscode-platformio) or submit the [issues](https://github.com/formulahendry/vscode-platformio/issues) to the GitHub Repo.

PlatformIO is really cool! Much better than Arduino IDE. 😄 Let's make PlatformIO better and better.

ivankravets, 2017-05-07:
Add project generator for VSCode with a support for:

* Code completion
* Multiple tasks/runners
* PIO Unified Debugger

dymo, 2017-05-10:
I have made pull requests for VS Code templates: https://github.com/platformio/platformio-core/pull/960 and https://github.com/platformio/platformio-docs/pull/4

I have been unable to get the PIO Unified Debugger to work, so I didn't make a template for the debug launch configuration.

ivankravets, 2017-05-27:
Hi guys,

We have a good news: **[PlatformIO IDE for VSCode](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide)**! 

*100% open source, maximum permissive Apache 2.0 license.*

Docs: http://docs.platformio.org/page/ide/vscode.html

[image]

This is a demo version, only base features are ported from PlatformIO IDE from Atom:

* Automatic Python + PIO Core Installer. No need to install Python or PIO Core before.
* Multiple Run Tasks
* PIO Unified Debugger for 100+ boards
* Automatic C/C++ Project Reindexer (when platformio.ini is changed or new libraries are installed)
* Integration with VSCode Internal Terminal.

Would be thankfufor any feedback. Please open issues in [PlatformIO IDE for VSCode repository](https://github.com/platformio/platformio-vscode-ide).

P.S: @formulahendry , thank you very much for your efforts on the own PlatformIO extension. Would you like to work on PlatformIO.Org extension? I propose to merge our efforts and make really cool exntension.

Petezah, 2017-05-28:
@ivankravets this sounds awesome! Can't wait to try it out.

formulahendry, 2017-05-29:
@ivankravets That's cool.

ivankravets, 2017-06-23:
Resolved with https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide

commit 64eaa151 (2017-03-26): Add information about VSCode integration // Resolve #619
closed 2017-06-23 by ivankravets