platformio/platformio-core #620: Support for MKR1000

gepd, 2016-04-15:
Will be SAMD Arduino/Genuino boards like MKR1000, supported soon or is possible to install it manually?

### Configuration

**Operating system**:

* OSX/Windows/Linux

### Additional info

Full information from arduino website: http://www.arduino.cc/en/Main/ArduinoMKR1000

sathibault, 2016-04-29:
I'm willing to contribute if there is some documentation on what needs to be done to add a board.

oxullo, 2016-06-19:
If I'm not mistaken, it can't be done without updating the arduinosam framework, adding the mkr1000 variant.
For what it matters, I added the board definition here: oxullo@6aac75d08d568123929770ceb285199cf3d30d32

valeros, 2016-06-21:
Please install PlatformIO [dev version](http://docs.platformio.org/en/latest/installation.html#development-version) and update `framework-arduinosam` package via `platformio update`

ivankravets, 2016-06-23:
@sathibault , @oxullo could you try the latest http://docs.platformio.org/en/latest/installation.html#development-version ? Need to understand that this issue is really fixed.

Thanks.

oxullo, 2016-06-26:
@valeros @ivankravets works for me! (green light for build and upload for the blink example plus the sketch I'm currently working on)

ivankravets, 2016-06-26:
@oxullo thanks a lot for the report!

closed 2016-06-26 by ivankravets