platformio/platformio-core #623: Undefined Reference Issue

djscheuf, 2016-04-18:
Hello PlatformIO,

Firstly, thank you for a great tool! I am running it through the Atom editor, with the platformio-de package v1.1.2 and platformio-ide-terminal v2.0.9 . I am working on a ControlledSerial implementation, which involves a CircularQueue, which I also developed. As for OS, I am running Linux Mint 17.2

Here is the error message: (The full output dump is included at the bottom of the post)
```
platformio run
[Mon Apr 18 17:58:15 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
--------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-
sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -DPLATFORMIO=020806 -I.pioenvs/uno
/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/infrastructure src/main.cpp
avr-g++ -o .pioenvs/uno/firmware.elf -Os -mmcu=atmega328p -Wl,--gc-sections,--relax .pioenvs/uno/src/main.o -L/home/daniel/.platfor
mio/packages/ldscripts -L.pioenvs/uno -Wl,--start-group .pioenvs/uno/libFrameworkArduinoVariant.a .pioenvs/uno/libFrameworkArduino.
a -lm .pioenvs/uno/libinfrastructure.a -Wl,--end-group

.pioenvs/uno/libinfrastructure.a(controlledSerial.o): In function `ControlledSerial::ControlledSerial()':
/home/daniel/Git/arduino-hackathon/project/.pioenvs/uno/infrastructure/util/controlledSerial.cpp:3: undefined reference to `CircularQueue<String>::CircularQueue()'
/home/daniel/Git/arduino-hackathon/project/.pioenvs/uno/infrastructure/util/controlledSerial.cpp:5: undefined reference to `CircularQueue<String>::CircularQueue(int)'
.pioenvs/uno/libinfrastructure.a(controlledSerial.o): In function `ControlledSerial::PostMessage(String)':
/home/daniel/Git/arduino-hackathon/project/.pioenvs/uno/infrastructure/util/controlledSerial.cpp:24: undefined reference to `CircularQueue<String>::Enque(String)'
.pioenvs/uno/libinfrastructure.a(controlledSerial.o): In function `ControlledSerial::PrintMessages()':
/home/daniel/Git/arduino-hackathon/project/.pioenvs/uno/infrastructure/util/controlledSerial.cpp:33: undefined reference to `CircularQueue<String>::HasItems()'
/home/daniel/Git/arduino-hackathon/project/.pioenvs/uno/infrastructure/util/controlledSerial.cpp:35: undefined reference to `CircularQueue<String>::Deque()'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/uno/firmware.elf] Error 1
========================= [ ERROR ] Took 0.45 seconds =========================
 
 PlatformIO: BuildFile 0Project 0No Issueslib/infrastructure/Data/circularQueue.hpp1:1
LFUTF-8C++Platfomio+291 update
```

under src/main.cpp:
```
#include <util/controlledSerial.hpp>

void setup()
{
  pSerial = new ControlledSerial();
  pSerial->PostMessage((String)"Setup executed: "+millis()/1000+"s.");
}

void loop()
{
  pSerial->PrintMessages();
}
```

Under lib, I have the following:
infrastructure
-Data
_--CircularQueue.hpp & .cpp_
-Interfaces
_--IControlledSerial.hpp_
-Utils
_--ControlledSerial.hpp & .cpp_

Here is the associated code:
**::CircularQueue.hpp**
```
#ifndef CIRCLEQUEUE
#define CIRCLEQUEUE

template <class T> class CircularQueue
{
  public:
    CircularQueue<T>(int size);
    CircularQueue<T>(); // default size = 10;

    virtual bool HasItems();

    virtual bool Enque(T item);
    virtual T Deque();

    virtual void Clear();
  private:
    virtual bool SizeLimitReached();
    virtual void EnqueAlgorithm(T item);
    virtual void IncrementFront();
    virtual void IncrementRear();

    int max;
    T* queue;
    int front;
    int rear;
};

#endif
```

**::IControlledSerial.hpp**
```
#ifndef ICONTROLLEDSERIAL
#define ICONTROLLEDSERIAL

#include <Arduino.h>

class IControlledSerial {
  public:
    virtual void Enable()=0;
    virtual void Disable()=0;

    virtual void PostMessage(String mesage)=0;

    virtual void PrintMessages()=0;
};

#endif
```

**::ControlledSerial.hpp**
```
#ifndef CONTROLLEDSERIAL
#define CONTROLLEDSERIAL

#include <Interfaces/iControlledSerial.hpp>
#include <pins.hpp>
#include <Data/circularQueue.hpp>

class ControlledSerial: public IControlledSerial
{
  public:
    ControlledSerial();

    void Enable();
    void Disable();

    void PostMessage(String message);

    void PrintMessages();
  private:
    CircularQueue<String> m_cQueue;
    bool m_bIsEnabled;
};

#endif
```

**::ControlledSerial.cpp**
```
#include "controlledSerial.hpp"

ControlledSerial::ControlledSerial()
{
  m_cQueue = CircularQueue<String>(20);
  m_bIsEnabled = true;
};

void ControlledSerial::Enable()
{
  m_bIsEnabled = true;
};

void ControlledSerial::Disable()
{
  m_bIsEnabled = false;
};

void ControlledSerial::PostMessage(String message)
{
  if(!m_bIsEnabled)
    return;

    m_cQueue.Enque(message);

};

void ControlledSerial::PrintMessages()
{
  if(!m_bIsEnabled)
    return;

    while(m_cQueue.HasItems())
    {
      Serial.println(m_cQueue.Deque());
    }
};
```

Now I have looked at a couple of other issue posts that are similar like #203 and #222. Sadly, neither of the solutions seem to work for me. As you can see my main is already a cpp, and most of my headers are .hpp to ensure the compiler understands I am trying to use C++.

Any suggestions or pointers would be greatly appreciated!

valeros, 2016-04-20:
Hi @djscheuf !
Could you provide please `CircularQueue.cpp` file as well?
You are using templates, so looks like you should put functions implementations in a [header file](http://stackoverflow.com/questions/495021/why-can-templates-only-be-implemented-in-the-header-file).

djscheuf, 2016-04-20:
Good Day @valeros,

Thank you for checking into this! I'd be happy to get you the contents of the CircularQueue.cpp when I get home from work, in about 8 or 9 hours. 

As to your observation, I am using templates, but was under the impression it would be wise to avoid implementation in the header file. If this is an oddity with templates, however, I'll try to move the implementation and try again; after I get you the .cpp contents.

Thank you for your help! And thank you for the reference!

djscheuf, 2016-04-20:
Thank you for your patience @valeros,

Here is the code for the CircularQueue.cpp
```
#include "circularQueue.hpp"

template<class T> CircularQueue<T>::CircularQueue(int size)
{
  max = size;
  queue = new T[max];
  front =-1;
  rear =-1;
}

template<class T> CircularQueue<T>::CircularQueue()
{
  max = 10;
  queue = new T[max];
  front =-1;
  rear =-1;
}

template<class T> bool CircularQueue<T>::HasItems()
{
  return (front != rear);
}

template<class T> bool CircularQueue<T>::Enque(T item)
{
  if(front ==-1)
  {
    front=0;
    rear=0;
    EnqueAlgorithm(item);
    return true;
  }
  else if(SizeLimitReached())
  {
    return false;
  }
  else{
    EnqueAlgorithm(item);
    return true;
  }
}

template<class T> void CircularQueue<T>::EnqueAlgorithm(T item)
{
  queue[rear] = item;
  IncrementRear();
}

template<class T> void CircularQueue<T>::IncrementRear()
{
  rear++;
  if(rear == max)
    rear = 0;
}

template<class T> bool CircularQueue<T>::SizeLimitReached()
{
  if(rear==front)
    return true;

  return false;
}

template<class T> T CircularQueue<T>::Deque()
{
  if(HasItems())
  {
    IncrementFront();
    return queue[front];
  }

  return 0;
}

template<class T> void CircularQueue<T>::IncrementFront()
{
  front++;
  if(front == max)
    front = 0;
}

template<class T> void CircularQueue<T>::Clear()
{
  front =-1;
  rear =-1;
}
```

I am going to try moving this implementation to the .hpp since that seems the most sustainable solution based upon your reference.

djscheuf, 2016-04-20:
@valeros , you've done it!

Thank you for your help! It was just the Template issue, thankfully nothing to do with PlatformIO! I apologize for taking your time, but thank you heartily for your help in resolving it!

The trick was to put the implementation in the .HPP file, and a minor correction in the Deque function of the Circular Queue. I am posting the correct version below for posterity. 

```
T Deque()
    {
      if(HasItems())
      {
        IncrementFront();
        return queue[front];
      }

      return T();
    };
```

Thank you again @valeros , your help and additional resources were , and are greatly appreciated!

closed 2016-04-20 by djscheuf