platformio/platformio-core #625: platformio serialports monitor does not exit properly on unconfigured serial port

blackketter, 2016-04-20:
### Configuration

**Operating system**:
MacOS X 10.11.4

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.8.6

### Description of problem
On MacOS, when using serial ports monitor and a USB target becomes disconnected (for example by uploading a Teensy3.1 firmware), an error is raised, but the command does not exit until the user hits a key.
This makes it difficult to automatically retry to reconnect.

#### Steps to Reproduce

1. monitor a USB device, for example: ```platformio serialports monitor -p /dev/tty.usbmodem935651```
2. upload a new firmware image in a separate console
3. exception is thrown and monitor waits for user input.

### Actual Results
This is the error output:

```Exception in thread Thread-2:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.macosx-10.10-intel/egg/serial/tools/miniterm.py", line 245, in reader
    data = character(self.serial.read(1))
  File "build/bdist.macosx-10.10-intel/egg/serial/serialposix.py", line 485, in read
    raise SerialException('read failed: %s' % (e,))
SerialException: read failed: [Errno 6] Device not configured
```

Then the user must hit a key to exit.

### Expected Results

The command should exit automatically without user input.

ivankravets, 2016-04-24:
PlatformIO uses PySerial's Miniterm. Please post this issue to https://github.com/pyserial/pyserial/issues

I don't have control on `SerialException: write failed: [Errno 6] Device not configured`, because PySerial stays in `while True` 😢 

P.S: Please add back link to this issue. I'll not close it. Thanks.

ivankravets, 2016-06-15:
Fixed in the latest PySerial. Please `pip install -U pyserial`

closed 2016-06-15 by ivankravets