platformio/platformio-core #626: Large Teensy 3.1 project build broken with large const data, works when built with Arduino/Teensyduino

blackketter, 2016-04-20:
------------------------------------------------------------------

### Configuration

**Operating system**:
MacOS X 10.11.4
Arduino 1.6.8, Teensyduino 1.228

**PlatformIO Version** (`platformio --version`):
2.8.6

### Description of problem
Large Teensy project builds but does not run correctly (some errors with i2c, etc.) when a largish constant data structure gets too big.

Builds and runs correctly under Arduino/Teensyduino

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env:teensy31]
lib-ignore = extras
platform = teensy
framework = arduino
board = teensy31
build_flags = -DTEENSY31 -UUSB_SERIAL -DTEENSYDUINO=127 -DARDUINO_ARCH_AVR -DUSB_SERIAL_HID -DLAYOUT_US_ENGLISH -I./src
board_f_cpu = 96000000
```

**Source file to reproduce issue:**
It's a large project.  The problem goes away when I reduce the size of one large const data structure from about 6k to 2k.  Happy to provide the full source if necessary.

Attached are the elf files and build logs for arduino and platformio

### Additional info
[arduino.buildresults.txt](https://github.com/platformio/platformio/files/229008/arduino.buildresults.txt)
[firmware.elfs.zip](https://github.com/platformio/platformio/files/229010/firmware.elfs.zip)
[platformio.buildresults.txt](https://github.com/platformio/platformio/files/229009/platformio.buildresults.txt)

valeros, 2016-04-21:
Hi @blackketter !
Could you provide an entire project, please? It would be easier to debug.
Thanks!

blackketter, 2016-04-22:
Thanks, @valeros! 

To make things easier, I pushed the project to github:  

https://github.com/blackketter/britepad.git

Use the platformio-broken branch to build with the large data structure, "keyConfig" in src/apps/CalculatorApp.cpp

The master branch has this file removed from the build, for now.

Let me know if there's anything I can do to help.

valeros, 2016-04-22:
Could you please try in turn these two variants:
1. Disable optimization, `build_flags = -O`
2. Try latest `arduinoteensy` package from [here](https://dl.dropboxusercontent.com/u/10025728/framework-arduinoteensy.zip) (just replace older version in PlatformIO home directory)

blackketter, 2016-04-22:
Aha!  -O seems to fix the issue, though I'm not sure what other side effects there might be.

valeros, 2016-04-28:
Hi @blackketter! Any news about your problem? Did `-O` optimization help?
I've just updated `teensy` framework to `1.28` version, so you can try it via `pio update`

blackketter, 2016-04-28:
Sorry, I wasn't clear.  -O did fix the problem.  It also increased the firmware image size, but that's the only side effect I noticed.  Looking back at the regular Arduino/Teensyduino build output, it did use -O as the optimization level, so I think this is the correct solution and should probably be the default optimization level.

ivankravets, 2016-06-15:
Could we close this issue?

closed 2016-06-17 by blackketter