platformio/platformio-core #628: Arduino Yun MIPS side support

pateldipen1984, 2016-04-21:
I wanted to develop some application for the arduino yun board especially mips side of it. I believe platformIO does not support it yet. Can some one take a look at it if it can be added in platformio? In a similar note, does platform IO support custom packages, I know we can create custom libraries, board and platform but not sure if I can do similar to adding packages? If it can, mips toolchain package can be added and it should be enough to develop mips side of the arduino yun.

ivankravets, 2016-07-16:
@artynet Do you have `toolchain-atmelavr` and `tool-avrdude` for MIPS?

Could you prepare these packages using the next file structure:

- https://bintray.com/platformio/dl-packages/tool-avrdude#files
- https://bintray.com/platformio/dl-packages/toolchain-atmelavr#files

Do we need specific hooks to flash Yun from Linino OS?

Thanks!

ivankravets, 2016-12-04:
@pateldipen1984 could you explain in details which packages do you need?

nopdotcom, 2017-05-09:
tl;dr: IMO embedded Linux MIPS binaries are a companion project, not platformio
-
(I say “OpenWrt" because it’s factory shipped on these particular devices, even though [LEDE](https://lede-project.org) is the live fork, and superior.)

I think @pateldipen1984 was asking for something similar to platformio that can generate OpenWrt binaries for the Yún, IMO, this is out of scope for platformio, since the MIPS sides of the Yún and LinkIt 7688 Smart/Smart Duo have reasonable/self-contained cross-compile toolchains.

Well, that’s my point of view. Given this bug report, they are obviously **not** reasonable toolchains for many people. It takes a lot of scaffolding to get `hello-world.c` built as a package. And then you have to link it, and then you have to transfer it to the Yún or LinkIt 7688. They could use an Arduino style of one button “compile this directory, deploy, and attach ‘serial’ console."

IMO dealing with the OpenWrt build system is a non-trivial problem for every adopter, but it is OpenWrt's problem. 

@ivankravets: If it’s really really a good idea, I might be able figure out how to wrap up the OpenWrt toolchain as platforms and toolchains. BTW, the LEDE folks have a bunch of improvements in 17.01—which brings up another problem. Which OpenWrt/LEDE target? There’s backwards compatibility for static binaries, but the dynamic libraries in the OpenWrt/LEDE distro can change with releases. 

Speaking of libraries, how does the platformio library system work with Linux? For dynamic library dependencies, the right thing is to do is to write a stub platformio library for each; it only needs to compile with `-I$OPENWRT_ROOT/include/foo` and link with `-foo`. If it’s a static dependency, it can be a real authentic platformio library, built from source.

The regular build product *probably* should be an `opkg` binary package, since it can state what it depends on. For iterative compile/test/debug cycles, it might be nice to deploy only the binary.

artynet, 2017-05-09:
@ivankravets : of course I have them both running and tested on OpenWrt (**uClibc** library) and LEDE (**musl** library). I needed more time to improve the toolchain cross-build and to run deep tests....I think it's time know to let them go "out in the woods".

We should make the same for the Tian since I am able to provide the ARM toolchain as well

@nopdotcom : I have cross-built the toolchains just for ar71xx architecture since our boards are relying on it but it's quite easy to have them running for another one. I have written a series of scripts taking care of it long time ago forking the original ones distributed with the AVR or ARM sources

ivankravets, 2018-06-18:
Moved to https://github.com/platformio/platform-atmelavr/issues/98

closed 2018-06-18 by ivankravets