platformio/platformio-core #629: avrdude version outdated

Xeranor, 2016-04-22:
### Configuration

**Operating system**: Arch Linux

**PlatformIO Version** (`platformio --version`): 2.8.6

### Description of problem
Shipped version of avrdude in tool-avrdude is 6.0.1 which needs ncurses 5.
Since ncurses 5 is out of date some compat libs are needed to run this version of avrdude.
The current version shipped with many distros is 6.3 which runs flawless. What speaks against updating it to the latest version?

valeros, 2016-04-22:
Thanks for the hint, but we are trying to use version shipped with Arduino IDE.
You can easily replace default package in `/home/user/.platformio/packages/tool-avrdude`

Xeranor, 2016-04-22:
Thank you for your fast reply. That seems logical - but shouldn't it be possible link the avrdude binary against local versions of libtinfo.so.5 and libncurses.5.so? 
Sure that wouldn't be a elegant solution, but it should make it work out of the box on most distributions.

valeros, 2016-04-22:
Maybe you are right, but we don't build binary, we use precompiled from Arduino IDE. We are going to update `avrdude` with the latest [package](http://downloads.arduino.cc/tools/avrdude-6.0.1-arduino5-x86_64-pc-linux-gnu.tar.bz2) does it have the same issue?

Xeranor, 2016-04-22:
Same problem with this package. I compiled libtinfo.5 and ncurses.5 and put them into the library folder - now it works! You can find it [https://www.dropbox.com/s/xfna7fwvz5dbqau/avrdude-6.0.1.tar.bz2?dl=0](here).

valeros, 2016-04-22:
PlatformIO 3.0 will support versioning packages via semver, thus, users will be able to choose the desired version of a package.

ivankravets, 2016-07-05:
`avrdude` has been updated. Please run `platformio update` command.

closed 2016-07-05 by ivankravets