platformio/platformio-core #631: Arduino: lack of -DUSB_MANUFACTURER makes manufacturer “Unknown"

nopdotcom, 2016-04-23:
**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.9.0.dev0

### Description of problem

The USB core for Arduino expects a USB manufacturer string as a preprocessor define, `-DUSB_MANUFACTURER`. See https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.cpp#L62 and https://github.com/adafruit/Adafruit_Arduino_Boards/blob/master/hardware/adafruit/avr/boards.txt#L26 for an example of where other packages define it.

#### Steps to Reproduce

I’m using a MediaTek LinkIt Smart 7688 DUO, but this should happen for anybody using their own VIDs.

1. Using pio, flash board with USB vendor ID other than Arduino LLC or SparkFun
2. (In IORegistryExplorer on OS X) “USB Vendor Name” now shows as “Unknown”
3. Upload a sketch using Arduino IDE
4. (In IORegistryExplorer on OS X) “USB Vendor Name” now shows as “MediaTek Labs"

### Additional info.

I solved this by appending to https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/frameworks/arduino.py#L126:
```
    if "usb_manufacturer" in BOARD_BUILDOPTS:
        ARDUINO_USBDEFINES.append('USB_MANUFACTURER=\\"%s\\"' % (env.subst(
            "${BOARD_OPTIONS['build']['usb_manufacturer']}").replace('"', "")))
```

This allows boards to specify a manufacturer if they care about it.

ivankravets, 2016-04-23:
@valeros please add `usb_manufacturer` for all AVR-based boards

ivankravets, 2016-04-23:
Please retest with the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-04-23 by ivankravets

nopdotcom, 2016-04-24:
This works for me.

ivankravets, 2016-04-24:
@nopdotcom thanks a lot for the report! Would be grateful for a star 🌟 https://github.com/platformio/platformio/stargazers

commit db0dbe6d (2016-04-23): Automatically add ``-DUSB_MANUFACTURER`` with vendor's name // Resolve #631