platformio/platformio-core #634: ESP8266 WifiUDP does not name a type

kissste, 2016-04-25:
I'm having trouble with compiling code for ESP8266 that send Udp messages.

#include <WifiUdp.h>
WiFiUDP Udp;

Gives an error: 'WifiUDP' does not name a type

The same code compiles well on Arduino's IDE target ESP8266

I have got the:
.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/WiFiUdp.cpp and .h
installed
```
Thank you

$ ~/.local/bin/platformio update

Platform espressif
--------
Updating toolchain-xtensa package:
Versions: Current=2, Latest=2    [Up-to-date]
Updating tool-esptool package:
Versions: Current=4, Latest=4    [Up-to-date]
Updating tool-mkspiffs package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=10, Latest=10          [Up-to-date]
Updating ldscripts package:
Versions: Current=19, Latest=19          [Up-to-date]

Updating [ 19 ] Adafruit-DHT library:
Versions: Current=edcd0e06e6, Latest=edcd0e06e6          [Up-to-date]
Updating [ 567 ] WifiManager library:
Versions: Current=0.11, Latest=0.11      [Up-to-date]
Updating [ 248 ] SoftwareSPI library:
Versions: Current=6f7e93171f, Latest=6f7e93171f          [Up-to-date]
Updating [ 539 ] SerialFlash library:
Versions: Current=682687ccb9, Latest=682687ccb9          [Up-to-date]
Updating [ 1101 ] ESP8266wifi library:
Versions: Current=f4c8072f1f, Latest=f4c8072f1f          [Up-to-date]

valeros, 2016-04-25:
Hi @kissste !
Looks like your sketch file has `*.c` extension instead of `*.cpp` or `.ino`.
Could you provide full build log, please?

kissste, 2016-04-25:
The src/main is .cpp

orangepi@OrangePI:~/PIRSK$ platformio run
[Mon Apr 25 09:12:28 2016] Processing esp12e (platform: espressif, board:
esp12e, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/src/main.o -c -fno-rtti
-fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals
-falign-functions=4 -ffunction-sections -fdata-sections -MMD
-DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266
-DARDUINO_ESP8266_ESP12 -DARDUINO=20100 -DPLATFORMIO=020806
-U__STRICT_ANSI__
-I/home/orangepi/.platformio/packages/framework-arduinoespressif/tools/sdk/include
-I.pioenvs/esp12e/FrameworkArduino
-I.pioenvs/esp12e/FrameworkArduinoVariant -I.pioenvs/esp12e/ESP8266WiFi
-I.pioenvs/esp12e/Adafruit_DHT_ID19 src/main.cpp
xtensa-lx106-elf-ar rcs .pioenvs/esp12e/libFrameworkArduinoVariant.a
xtensa-lx106-elf-ranlib .pioenvs/esp12e/libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/FrameworkArduino/Esp.o -c -fno-rtti
-fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals
-falign-functions=4 -ffunction-sections -fdata-sections -MMD
-DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266
-DARDUINO_ESP8266_ESP12 -DARDUINO=20100 -U__STRICT_ANSI__
-I/home/orangepi/.platformio/packages/framework-arduinoespressif/tools/sdk/include
-I.pioenvs/esp12e/FrameworkArduino
-I.pioenvs/esp12e/FrameworkArduinoVariant
.pioenvs/esp12e/FrameworkArduino/Esp.cpp
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/FrameworkArduino/FS.o -c -fno-rtti
-fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals
-falign-functions=4 -ffunction-sections -fdata-sections -MMD
-DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266
-DARDUINO_ESP8266_ESP12 -DARDUINO=20100 -U__STRICT_ANSI__
-I/home/orangepi/.platformio/packages/framework-arduinoespressif/tools/sdk/include
-I.pioenvs/esp12e/FrameworkArduino
-I.pioenvs/esp12e/FrameworkArduinoVariant
.pioenvs/esp12e/FrameworkArduino/FS.cpp
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/FrameworkArduino/HardwareSerial.o
-c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls
-mtext-section-literals -falign-functions=4 -ffunction-sections
-fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266
-DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=20100
-U__STRICT_ANSI__
-I/home/orangepi/.platformio/packages/framework-arduinoespressif/tools/sdk/include
-I.pioenvs/esp12e/FrameworkArduino
-I.pioenvs/esp12e/FrameworkArduinoVariant
.pioenvs/esp12e/FrameworkArduino/HardwareSerial.cpp
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/FrameworkArduino/IPAddress.o -c
-fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls
-mtext-section-literals -falign-functions=4 -ffunction-sections
-fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266
-DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=20100
-U__STRICT_ANSI__
-I/home/orangepi/.platformio/packages/framework-arduinoespressif/tools/sdk/include
-I.pioenvs/esp12e/FrameworkArduino
-I.pioenvs/esp12e/FrameworkArduinoVariant
.pioenvs/esp12e/FrameworkArduino/IPAddress.cpp
src/main.cpp:11:1: error: 'WifiUDP' does not name a type
WifiUDP Udp;
^

Thank you very much

valeros, 2016-04-25:
Could you show your sketch?
I've just compiled simple [example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino) with UDP socket and everything works fine.

kissste, 2016-04-25:
Thank you very much, your example UDP compiled with no errors.

Problem was on my side with my sketch, as simple as lower/upper case
WifiUDP vs WiFiUDP.

Thank you again

closed 2016-04-25 by valeros