platformio/platformio-core #636: Add support for ATtiny13

functionpointer, 2016-04-26:
As requested by ivankravets on the forum (https://community.platformio.org/t/upload-to-attiny13/198) i request support for generic ATtiny13-based boards. After the addition of ATTiny24, ATTiny25, ATTiny45 and ATTiny85 in PlatformIO 2.9.0 and an existing arduino core for the ATtiny13 (https://sourceforge.net/projects/ard-core13/) this is a logical step.

ivankravets, 2016-04-28:
Please re-test the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-04-28 by ivankravets

functionpointer, 2016-04-28:
Unfortunately it seems that the arduino files for the attiny13 are missing:
```
# platformio run
[04/28/16 15:24:52] Processing attiny (upload_protocol: usbasp, upload_flags: -Pusb, board: attiny13, framework: arduino, platform: atmelavr)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-ar rcs .pioenvs\attiny\libFrameworkArduino.a
avr-g++ -o .pioenvs\attiny\src\main.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=attiny13 -DF_CPU=9600000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_ATTINY13 -DARDUINO=10608 -DPLA
TFORMIO=020900 -I.pioenvs\attiny\FrameworkArduino src\main.cpp
src\main.cpp:1:21: fatal error: Arduino.h: No such file or directory
#include "Arduino.h"
^
compilation terminated.
scons: *** [.pioenvs\attiny\src\main.o] Error 1
======================================================================================================== [ ERROR ] Took 0.37 seconds ========================================================================================================

```

Programs for the ATtiny24 and 85 are compiling. After inspecting the changes in commit  #0cf78c4 i found that the ATtiny was already set up for core13, but the relevant folder in ~\.platformio\packages\framework-arduinoavr\cores was missing. After copying them in i was able to compile programs for the ATtiny13.
This would be fixed by including the core13 folder from https://sourceforge.net/projects/ard-core13/ into ~/.platformio/packages/framework-arduinoavr/cores/

Uploading also has a problem:
```
"C:\Users\whatever\.platformio\packages\tool-avrdude\avrdude" -v -p attiny13 -C "C:\Users\whatever\.platformio\packages\tool-avrdude\avrdude.conf" -c usbasp -Pusb -D -U flash:w:.pioenvs\attiny\firmware.hex:i
[...]
Programmer Type : usbasp
Description     : USBasp, http://www.fischl.de/usbasp/
 
avrdude: AVR device initialized and ready to accept instructions
 
Reading | ################################################## | 100% 0.00s

 
avrdude: Device signature = 0x1e9007
avrdude: reading input file ".pioenvs\attiny\firmware.hex"
avrdude: writing flash (350 bytes):
 
Writing | ################################################## | 100% 0.25s
 
avrdude: 350 bytes of flash written
avrdude: verifying flash memory against .pioenvs\attiny\firmware.hex:
avrdude: load data flash data from input file .pioenvs\attiny\firmware.hex:
avrdude: input file .pioenvs\attiny\firmware.hex contains 350 bytes
avrdude: reading on-chip flash data:
 
Reading | ################################################## | 100% 0.18s
 
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0006
avrdude: verification error; content mismatch
avrdude done.  Thank you.
0x20 != 0x28
```

After the writing the new program to the flash storage avrdude reads it again to verify that the program was correctly uploaded. Since avrdude gets called with the -D option it does not erase the chip before writing to the flash storage. This results in a failed verification and upload (ATtiny doesnt run either old nor new program). Manually calling avrdude from the command line without the -D parameter works. This also shows that the generated firmware.hex file is valid and runnable on the ATtiny13.

ivankravets, 2016-04-28:
Did you run `platformio update`?

functionpointer, 2016-04-28:
I am using the IDE, and yes i did upgrade platformio and i switched to the development version as stated in the documentation. How else would i have all the attiny entries (including attiny13) when running pio boards?

ivankravets, 2016-04-28:
1. Give me please an output of this command `pio platforms show atmelavr`
2. Use `pio run -t program` instead `pio run -t upload` (Upload using Programmer)

The version of framework should be 29.

functionpointer, 2016-04-28:
Well, `pio platforms show atmelavr` told me i was using framework version 28. Evidently i did not update correctly. Running `pio update` as you suggested fixed that and downloaded core13. The first time i used the menu entry "Upgrade PlatformIO IDE" which apparently did not work even though it told me it did.

when using `pio run -t upload` the upload works.

In conclusion, everything appears to be working. Thank you ivan!

ivankravets, 2016-04-28:
Don't forget to update to stable release :) PlatformIO IDE 1.2.0 and CLI 2.9.0 is going to be released today.

commit 0cf78c42 (2016-04-28): Added support for Generic ATTiny13 // Resolve #636