platformio/platformio-core #637: Error compiling sketch that works in Arduino IDE

mplewis, 2016-04-28:
- [x] Bug report. If you’ve found a bug, please provide information below.

------------------------------------------------------------------

This sketch compiles fine in Arduino IDE 1.6.8 with Teensyduino 1.28 for Teensy 3.0.

### Configuration

**Operating system**: OS X 10.11.4

**PlatformIO Version** (`platformio --version`): 2.8.6

### Description of problem

#### Steps to Reproduce

Compile the attached sketch.

### Actual Results

The following errors are thrown:

```
platformio run --target upload
[Wed Apr 27 19:17:23 2016] Processing env (platform: teensy, board: teensy30, framework: arduino)
------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/env/src/tmp_ino_to.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -DPLATFORMIO=020806 -I.pioenvs/env/FrameworkArduino src/tmp_ino_to.cpp
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/Print.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/Print.cpp
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/Stream.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/Stream.cpp
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/Tone.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/Tone.cpp
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/WMath.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/WMath.cpp
src/tmp_ino_to.cpp:5:17: error: 'Location' was not declared in this scope
double distance(Location from, Location to);
^
src/tmp_ino_to.cpp:5:32: error: 'Location' was not declared in this scope
double distance(Location from, Location to);
^
src/tmp_ino_to.cpp:5:43: error: expression list treated as compound expression in initializer [-fpermissive]
double distance(Location from, Location to);
^
src.ino: In function 'double distance(Location, Location)':
src.ino:9:43: error: 'double distance(Location, Location)' redeclared as different kind of symbol
src/tmp_ino_to.cpp:5:8: error: previous declaration of 'double distance'
double distance(Location from, Location to);
^
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/WString.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/WString.cpp
arm-none-eabi-gcc -o .pioenvs/env/FrameworkArduino/analog.o -c -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/analog.c
scons: *** [.pioenvs/env/src/tmp_ino_to.o] Error 1
arm-none-eabi-g++ -o .pioenvs/env/FrameworkArduino/avr_emulation.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=48000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX128__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/env/FrameworkArduino .pioenvs/env/FrameworkArduino/avr_emulation.cpp
========================================== [ ERROR ] Took 0.91 seconds ==========================================
```

### Expected Results

Sketch compiles successfully.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:env]
platform = teensy
framework = arduino
board = teensy30
```

**Source file to reproduce issue:**
```cpp
#include <math.h>

typedef struct Location {
  char name[16];
  double lat;
  double lng;
} Location;

double distance(Location from, Location to) {
  return sqrt(pow(from.lat - to.lat, 2) + pow(from.lng - to.lng, 2));
}

void setup() {
}

void loop() {
}
```

ivankravets, 2016-04-28:
I've rewritten INO to CPP converter in 2.9.0. Please try the latest development version. Does it work?

ivankravets, 2016-04-28:
Resolved in https://github.com/platformio/platformio/commit/2682e8db8adbbdefba83aeb3123e92e3eac87a0d

closed 2016-04-28 by ivankravets

mplewis, 2016-04-28:
Looks great now. Thanks for the quick response! 🍪