platformio/platformio-core #639: Build error after the "Big Update"

Sufomba, 2016-04-29:
Hi all,

Before the 'big update' everything was fine, after the 'big update' this error is happening.
The library 'TimedAction' is inside the lib folder.

Atom : 1.7.3
Electron: 0.36.8
Chrome : 47.0.2526.110
Node : 5.1.1

Linux Mint 17.3

Code:
---------
#include <Arduino.h>
#include <TimedAction.h>
//this initializes a TimedAction class that will change the state of an LED every second.
TimedAction checapassoutempo = TimedAction(1000,blink);
//pin / state variables
const byte ledPin = 13;
boolean ledState = false;

void setup(){
  pinMode(ledPin,OUTPUT);
  digitalWrite(ledPin,ledState);
}

void loop(){
  checapassoutempo.check();
  //do something else
}

void blink(){
 ledState ? ledState=false : ledState=true;
 digitalWrite(ledPin,ledState);
}
--------

>>>> Error: 'blink' was not declared in this scope <<<<

Fix:
By ivankravetsThe PlatformIO Team
Temporary rename timed_act.ino to timed_act.cpp
Put after all #include this "void blink();" 

Thank you guys.

closed 2016-04-30 by ivankravets

alexgorbatchev, 2017-08-06:
Same issue on 3.5.0a1

ivankravets, 2017-09-20:
@alexgorbatchev have you resolved your issue?

alexgorbatchev, 2017-09-20:
Yes, thank you!

commit a77cbe5c (2016-04-30): Handle pointer to the prototypes while converting "*.ino" to ".cpp" // Resolve #639