platformio/platformio-core #640: Support Mbed OS (3.x) with yotta build

georgeedwards, 2016-04-29:
What kind of issue is this?

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

As you may be aware, the mbed team are in the final stages of rolling out the yotta build engine to support the new Mbed OS system of building applications for their hardware. 

As far as I know, this isn't supported in platformo currently (I would love to be corrected on this). If that is the case, are there any plans to support it, if not - can we make some?

ivankravets, 2016-04-29:
Sorry for the off-top. I see request https://raw.githubusercontent.com/Gas-Sense/ble/master/library.json

Does this lib work with PlatformIO?

georgeedwards, 2016-04-29:
@ivankravets yes, I have tested it manually moving it into the project and it works well!

ivankravets, 2016-04-30:
The `BLE_API` library has been added. http://platformio.org/lib/show/296/BLE_API

georgeedwards, 2016-04-30:
@ivankravets Thanks! Any thougths on yotta?

ivankravets, 2016-04-30:
PlatformIO uses own Build System that works everywhere without dependences. `yotta` depends on CMake and Ninja Build System. You will have ability in PlatformIO 3.0 to write own development platform and replace our Build System with `yotta`.

The best option here is to write native PlatformIO System Build script for Mbed OS. We will try to find sponsors for this feature because we don't have enough resources for 3-rd parties tools.

ivankravets, 2016-08-26:
@valeros , please switch all development platforms to new builder for ARM mbed and close this issue.

ivankravets, 2016-08-31:
PlatformIO 3.0 supports to build Mbed OS

closed 2016-08-31 by ivankravets