platformio/platformio-core #641: Mbed framework multiple definition on FRDM-KL46Z

raymore, 2016-04-30:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------
At a certain point in the development of my project all sources build, but the linkage fails with multiple definitions of : `__gnu_cxx::__verbose_terminate_handler()'.
The culprits are : frdm_kl46z/libFrameworkMbed.a(retarget.o) and arm-none-eabi/lib/armv6-m/libstdc++_s.a(vterminate.o):vterminate.cc

This seems to happen when the project gets to a certain size or number of files, as I have built and run this project for debugging for many iterations until this happened. Rolling back the recent changes to the code does not undo the outcome.

ini:
[env:frdm_kl46z]
platform = freescalekinetis
framework = mbed
board = frdm_kl46z
build_flags = -std=c++11

This is the second platformio/atom project I have had this issue with. The first is on hold.
Running:
Platformio IDE 1.2.0, CLI 2.9.0

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

valeros, 2016-05-03:
Hi @raymore !
Could you provide an example to reproduce this issue, please?

raymore, 2016-05-03:
Hi,

Nice to hear from you. As mention in the report it wasn't until I had 
been developing for some time that the issue surfaced. I now have a 
project of over 150K of source files. I'll see if I can cut the code 
down to a managable size to upload that still demonstrates the issue.

It seems very similar to the STM32 issue, i.e. reports the same message. 
Is it your policy to only fix issues for the reported platform or do you 
consider a likelyhood of an issue similar to mine on a FRDM-KL46Z being 
affected by duplicates as well (if it involves the same libraries)?

I'll work in a smaller example and let you know.

raymore, 2016-05-03:
Hi,

I have savagely cut most of the code that does not impact the issue and 
have discovered that the culprit is a pure virtual function in 
StateMachine.h. Removing the function and it compiles OK. Restoring it 
and the issue returns.

I am at a loss to provide an insight into why I get this linkage failure:
.pioenvs/frdm_kl46z/libFrameworkMbed.a(retarget.o): In function `__gnu_cxx::__verbose_terminate_handler()':
retarget.cpp:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): multiple definition of `__gnu_cxx::__verbose_terminate_handler()'
/home/ray/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv6-m/libstdc++_s.a(vtermina
te.o):vterminate.cc:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/frdm_kl46z/firmware.elf] Error 1

Please find the code attached.

Regards

valeros, 2016-05-04:
Hi @raymore !
Sorry, but I can't find any attached file. Could you share your project as zip file somewhere?

raymore, 2016-05-04:
Hi,
I'll try again as I do not have share presence on the web.

I can see that I have attached the zip file correctly, so let me know if 
you still have difficulty accessing it.

Regards

valeros, 2016-05-04:
Please send me your project directly to `vkoval@platformio.org`
Thanks!

valeros, 2016-05-06:
Please install PlatformIO from the special branch:
`pip install https://github.com/platformio/platformio/archive/feature/mbed-build-error-issue-641.zip`
Does it work well with your project?
Thanks!

raymore, 2016-05-07:
Hi,
I have upgraded as per your email. I ran platformio in atom and the 
result still raised the issue, so I suspect that the platformio/atom 
build is using some of the old system. I then ran 'pio run' on my 
original project (non cut down) and it built OK.

I may have to re-install atom.

As a developer, I am curious about the solution. Can you give some 
details regarding the fix?

Regards

orgkhnargh, 2016-05-07:
@raymore 
Did you run `pip install https://github.com/platformio/platformio/archive/feature/mbed-build-error-issue-641.zip` from inside of the IDE terminal?

raymore, 2016-05-07:
Hi,
I have installed the suggested zip file from the command line and the 
project compiles and links as it should. I have just installed from the 
terminal inside atom but the same error still exists.

The command line: pio run, still works.

Regards

ivankravets, 2016-05-07:
@raymore give me please an output of `pio --version` command. Please run it in PlatformIO IDE Terminal.

raymore, 2016-05-08:
Hi,
The pio --version (in the atom IDE terminal) is: PlatformIO, version 
2.9.2.dev0

I have just executed: pio run (in the atom IDE terminal) and the build 
now completes, so the install in the terminal required a reboot.

But if I execute the build from the IDE panel button it fails!!!

I have attached the output of the build.

Regards

[Sun May  8 08:41:59 2016] Processing frdm_kl46z (platform: freescalekinetis, board: frdm_kl46z, framework: mbed, build_flags: -std=c++11)
--------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/frdm_kl46z/src/main.o -c -std=gnu++98 -fno-rtti -std=c++11 -fdata-sections -ffunction-sections -Wno-unused-parameter -mcp
u=cortex-m0plus -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2
-fomit-frame-pointer -Wall -MMD -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_KLXX -DTARGET_CORTEX_M -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_Freesc
ale -DTOOLCHAIN_GCC -D__MBED__=1 -D__CORTEX_M0PLUS -DMBED_BUILD_TIMESTAMP=1459422030.97 -DTARGET_KL46Z -DTARGET_LIKE_CORTEX_M0 -DARM_MATH_CM0PLUS -DPLA
TFORMIO=020900 -I.pioenvs/frdm_kl46z/FrameworkMbedInc248832578 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-213564679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-
250686161 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-584841671 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-542615144 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-1840
941679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc709138787 -I.pioenvs/frdm_kl46z/FrameworkMbedInc1550639565 -I.pioenvs/frdm_kl46z/Util -I.pioenvs/frdm_kl46
941679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc709138787 -I.pioenvs/frdm_kl46z/FrameworkMbedInc1550639565 -I.pioenvs/frdm_kl46z/Util -I.pioenvs/frdm_kl46
z/CommsManager src/main.cpp
arm-none-eabi-ar rcs .pioenvs/frdm_kl46z/libFrameworkMbed.a /home/ray/.platformio/packages/framework-mbed/variant/KL46Z/mbed/TARGET_KL46Z/TOOLCHAIN_GCC
_ARM/cmsis_nvic.o /home/ray/.platformio/packages/framework-mbed/variant/KL46Z/mbed/TARGET_KL46Z/TOOLCHAIN_GCC_ARM/mbed_overrides.o /home/ray/.platformi
o/packages/framework-mbed/variant/KL46Z/mbed/TARGET_KL46Z/TOOLCHAIN_GCC_ARM/retarget.o /home/ray/.platformio/packages/framework-mbed/variant/KL46Z/mbed
/TARGET_KL46Z/TOOLCHAIN_GCC_ARM/startup_MKL46Z4.o /home/ray/.platformio/packages/framework-mbed/variant/KL46Z/mbed/TARGET_KL46Z/TOOLCHAIN_GCC_ARM/board
.o /home/ray/.platformio/packages/framework-mbed/variant/KL46Z/mbed/TARGET_KL46Z/TOOLCHAIN_GCC_ARM/system_MKL46Z4.o
arm-none-eabi-ranlib .pioenvs/frdm_kl46z/libFrameworkMbed.a
arm-none-eabi-g++ -o .pioenvs/frdm_kl46z/Util/Util.o -c -std=gnu++98 -fno-rtti -std=c++11 -fdata-sections -ffunction-sections -Wno-unused-parameter -mc
pu=cortex-m0plus -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2
 -fomit-frame-pointer -Wall -MMD -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_KLXX -DTARGET_CORTEX_M -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_Frees
cale -DTOOLCHAIN_GCC -D__MBED__=1 -D__CORTEX_M0PLUS -DMBED_BUILD_TIMESTAMP=1459422030.97 -DTARGET_KL46Z -DTARGET_LIKE_CORTEX_M0 -DARM_MATH_CM0PLUS -I.p
ioenvs/frdm_kl46z/FrameworkMbedInc248832578 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-213564679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-250686161 -I.pioenv
s/frdm_kl46z/FrameworkMbedInc-584841671 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-542615144 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-1840941679 -I.pioenvs/f
rdm_kl46z/FrameworkMbedInc709138787 -I.pioenvs/frdm_kl46z/FrameworkMbedInc1550639565 -I.pioenvs/frdm_kl46z/Util -I.pioenvs/frdm_kl46z/CommsManager .pio
envs/frdm_kl46z/Util/Util.cpp
arm-none-eabi-g++ -o .pioenvs/frdm_kl46z/CommsManager/CommsManager.o -c -std=gnu++98 -fno-rtti -std=c++11 -fdata-sections -ffunction-sections -Wno-unus
ed-parameter -mcpu=cortex-m0plus -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c
-fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_KLXX -DTARGET_CORTEX_M -DTARGET_LIKE_MBED -DTARGET_M0
P -DTARGET_Freescale -DTOOLCHAIN_GCC -D__MBED__=1 -D__CORTEX_M0PLUS -DMBED_BUILD_TIMESTAMP=1459422030.97 -DTARGET_KL46Z -DTARGET_LIKE_CORTEX_M0 -DARM_M
ATH_CM0PLUS -I.pioenvs/frdm_kl46z/FrameworkMbedInc248832578 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-213564679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-250
686161 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-584841671 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-542615144 -I.pioenvs/frdm_kl46z/FrameworkMbedInc-1840941
679 -I.pioenvs/frdm_kl46z/FrameworkMbedInc709138787 -I.pioenvs/frdm_kl46z/FrameworkMbedInc1550639565 -I.pioenvs/frdm_kl46z/Util -I.pioenvs/frdm_kl46z/C
ommsManager .pioenvs/frdm_kl46z/CommsManager/CommsManager.cpp
arm-none-eabi-ar rcs .pioenvs/frdm_kl46z/libUtil.a .pioenvs/frdm_kl46z/Util/Util.o
arm-none-eabi-g++ -o .pioenvs/frdm_kl46z/CommsManager/StateMachine.o -c -std=gnu++98 -fno-rtti -std=c++11 -fdata-sections -ffunction-sections -Wno-unus
ed-parameter -mcpu=cortex-m0plus -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c
-fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_KLXX -DTARGET_CORTEX_M -DTARGET_LIKE_MBED -DTARGET_M0
P -DTARGET_Freescale -DTOOLCHAIN_GCC -D__MBED__=1 -D__CORTEX_M0PLUS -DMBED_BUILD_TIMESTAMP=1459422030.97 -DTARGET_KL46Z -DTARGET_LIKE_CORTEX_M0 -DARM_M
[... cut ...]
retarget.cpp:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): multiple definition of `__gnu_cxx::__verbose_terminate_handler()'
/home/ray/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv6-m/libstdc++_s.a(vtermina
te.o):vterminate.cc:(.text._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/frdm_kl46z/firmware.elf] Error 1
========================= [ ERROR ] Took 0.46 seconds =========================

valeros, 2016-06-02:
Hi @raymore !
Please re-test your project with the latest PlatformIO from the [develop](http://docs.platformio.org/en/latest/installation.html#development-version) branch.
Thanks!

commit 38967eab (2016-06-02): Fix issue with ARM mbed framework and multiple definition errors   on FRDM-KL46Z board // Resolve #641
commit 06e42dc6 (2016-06-02): Merge branch 'develop' into feature/platformio-30

* develop:
  Version bump to 2.9.2 (issues #641, #645, #648, #652, #664, #665, #666, #671, #674)
  Fix issue with ARM mbed framework when abstract class breaks compile for LPC1768 // Resolve #666
  Fix issue with ARM mbed framework and multiple definition errors   on FRDM-KL46Z board // Resolve #641
  List embedded boards in docs
  Fix multiple definition in mbed framework when using abstract class // Issue #641, #666
  Add "stlink" as the default uploader for STM32 Discovery boards // Resolve #665
  Implement grep serial ports for Windows
  R [... cut]
closed 2016-06-02 by ivankravets