platformio/platformio-core #642: CLI 2.9.0 fatal bug:

seclorum, 2016-04-30:
Updated to CLI 2.9.0 this evening, and now this happens - a real productivity killer, since it seems to originate deep in some other header, somewhere.  I still haven't found a fix...

````/Users/seclorum/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
.pioenvs/esp12e/FrameworkArduino/Arduino.h:254:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))
````

I have to say, I'm getting less and less inclined to upgrade PlatformIO, and more and more inclined to just switch back to the Espressif Toolchain, with every upgrade.  The last two upgrades to PIO have been show-stoppers and very counter-productive to my project ... may I suggest a few less 'cool' features, and a lot more writing of tests, in future releases .. for that matter, does  PIO even have testing?

ivankravets, 2016-05-01:
> I have to say, I'm getting less and less inclined to upgrade PlatformIO, and more and more inclined to just switch back to the Espressif Toolchain, with every upgrade.

We highly recommend upgrading to each new release. We haven't changed PlatformIO CLI since the first 0.1.0 release. I understand you. Please sorry for the inconvenience. However, please don't mix PlatformIO CLI & Espressif. PlatformIO is Build System, Library Manager and etc., but we don't have relation to 3-rd party software. 

Some users contact us and ask to update Arduino Espressif Framework to the latest release and we do it. As a result, other users have problems. If you use PlatformIO for hours and hours and should take notice that it is a very stable ecosystem. I don't remember when we have had big problems with upgrades.

Nevertheless, I don't defend PlatformIO. I know about this possible conflict between upgrades. In this case, I opened separate issue https://github.com/platformio/platformio/issues/479. PlatformIO 3.0 will split "PlaformIO" from "Vendors related software". PlatformIO users will decide which version of a development platform to use.

> for that matter, does PIO even have testing?

Of course, each new commit to PlatformIO CLI is covered with a lot of different examples https://github.com/platformio/platformio-examples. See 

* https://travis-ci.org/platformio/platformio/builds (Unix)
* https://ci.appveyor.com/project/ivankravets/platformio/history (Windows)

-----------

Could you provide "Configuration" information how to reproduce this issue? See https://github.com/platformio/platformio/blob/develop/.github/ISSUE_TEMPLATE.md

Thanks.

seclorum, 2016-05-01:
````
$ pio --version
PlatformIO, version 2.9.1
````

The last two updates have been catastrophic for me, in terms of productivity lost.  The 2.8.x (I forget the minor version - it happened exactly a week ago) upgrade shipped with a broken "common.h" in framework-arduinoespressif/variants/generic, which I 'fixed' by commenting out the offending lines, so I could continue working.  That shook my faith.

And now yesterdays upgrade seems to have introduced some kind of bug with the max() macro, which appears to be defined elsewhere as an inline method, though I have not discovered the root of the problem - I simply commented out the macro definition in Arduino.h (see above path) so I could continue working on my project.

Alas I don't have access to previous PIO installations, or else I'd dig deeper - but you can reproduce these problems by trying to build my project, if you have the ability to revert to 2.8.x (I don't.)

>for that matter, does PIO even have testing?

Well, I looked at the test, and its a very minor test of the espressif SDK, and does not actually test the C++ compiler functionality, since its a single .ino file, no .h/.cpp files.  In my case, I'm building a C++ project that is a heck of a lot more complex than the example, and you are welcome to see the code for yourself, since its open source:  http://github.com/magicshifter/MS3000-Firmware/

I apologize for not being able to produce more detailed results - at the moment I have to stay focused on my work-related issues, but I will attempt to resolve this with you if you can spend time on it.

ivankravets, 2016-05-01:
I've just re-tested your project with PlatformIO 2.8 and 2.9. The problem is not with PlatformIO. Please switch temporarily to the previous `2.1.0` version of @esp8266 framework. See docs http://docs.platformio.org/en/latest/platforms/espressif.html#using-arduino-framework-with-staging-version

Here is the tag https://github.com/esp8266/Arduino/tree/2.1.0

P.S: If you need to downgrade PlatformIO, just use `pip install -U "platformio<2.9"`
P.S.S: Do you have problem with Espressif for Arduino IDE and 2.2.0 version? Looks, like need to report this issue here https://github.com/esp8266/Arduino/issues

ivankravets, 2016-05-01:
> Well, I looked at the test, and its a very minor test of the espressif SDK, and does not actually test the C++ compiler functionality, since its a single .ino file, no .h/.cpp files.

It is more complicated test than native C/C++ source code because it requires `INO` to `CPP` converter.

In any case, I will think about adding tests to the open source projects to existing CI.

seclorum, 2016-05-01:
Awesome, thanks for looking into this ivankravets - I will do as you suggest and try to find the proper place to report this bug.  I apologize for the incorrect assignment of the problem to PIO - I hope we will leave this issue though, for others who may run into this wall as well.

ivankravets, 2016-05-03:
See answer by author ( @igrr ) of Arduino for ESP8266 https://github.com/platformio/platformio/issues/646#issuecomment-216244875

closed 2016-05-03 by ivankravets

seclorum, 2016-05-03:
Thanks for that @ivankravets  .. I can confirm that this workaround fixed it.