platformio/platformio-core #645: SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1754)

zindy, 2016-05-02:
What kind of issue is this?

- **Bug report.** If you’ve found a bug, please provide information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 16.04 running as an LXC container on Proxmox 4.1-1 (LXC allocation: 2GB RAM / 1GB SWAP, CPU limit:1, CPU units:1024, 40GB space for the root disk)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.9.1

### Description of problem

Trying to compile the blink example for a nodemcu by running sudo platformio run as per http://docs.platformio.org/en/latest/quickstart.html

#### Steps to Reproduce

1. `sudo platformio run`

[Mon May  2 11:48:45 2016] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
The platform 'espressif' has not been installed yet. Would you like to install it now? [y/N]: 

2. press y

### Actual Results

```
Installing tool-scons package:
Already installed
Installing toolchain-xtensa package:
Downloading
Downloading
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 97, in cli
    results.append(ep.process())
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 133, in process
    result = self._run()
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/run.py", line 199, in _run
    return p.run(build_vars, build_targets, self.verbose_level)
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 453, in run
    self._install_default_packages()
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 439, in _install_default_packages
    return self.install()
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 356, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 102, in install
    dlpath = self.download(info['url'], pkg_dir, info['sha1'])
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 45, in download
    fd.start()
  File "/usr/local/lib/python2.7/dist-packages/platformio/downloader.py", line 76, in start
    f.write(next(itercontent))
  File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 664, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/response.py", line 353, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/response.py", line 310, in read
    data = self._fp.read(amt)
  File "/usr/lib/python2.7/httplib.py", line 612, in read
    s = self.fp.read(amt)
  File "/usr/lib/python2.7/socket.py", line 384, in read
    data = self._sock.recv(left)
  File "/usr/lib/python2.7/ssl.py", line 752, in recv
    return self.read(buflen)
  File "/usr/lib/python2.7/ssl.py", line 639, in read
    v = self._sslobj.read(len)
SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1754)
```

### Expected Results

Blinking lights maybe?

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:nodemcu]
platform = espressif
framework = arduino
board = nodemcu

[env:nodemcuv2]
platform = espressif
framework = arduino
board = nodemcuv2
```

**Source file to reproduce issue:**
```cpp
/**
 * Blink
 *
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */
#include "Arduino.h"

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(1000);
}
```

### Additional info

That's all I can think of. Ubuntu is up and running and was recently updated. So is Proxmox.

ivankravets, 2016-05-03:
See https://www.google.com.ua/search?q=ubuntu+ssl+DECRYPTION_FAILED_OR_BAD_RECORD_MAC

Could you try to upgrade OpenSSL and root certificates?

zindy, 2016-05-03:
Hi Ivan,

thank you so much for pointing me in the right direction.

As per https://www.brightbox.com/blog/2014/03/04/add-cacert-ubuntu-debian/

I issued the following commands:
```
sudo mkdir /usr/local/share/ca-certificates/cacert.org
sudo wget -P /usr/local/share/ca-certificates/cacert.org http://www.cacert.org/certs/root.crt http://www.cacert.org/certs/class3.crt
sudo update-ca-certificates
```

The last one took a bit longer but then I was able to:
```
sudo platformio run
```

and after a while was greeted with:
```
218486      884   29480  248850   3cc12 .pioenvs/nodemcuv2/firmware.elf
========================= [SUCCESS] Took 10.46 seconds =========================
```

Cheers!

**PS:** One for the troubleshooting guide? I could write that up if it helps.

zindy, 2016-05-03:
I may have been a bit overoptimistic on that one. On the same machine (with updated root certificates) I just tried a
```
wget https://grafanarel.s3.amazonaws.com/builds/grafana_3.0.0-beta71462173753_amd64.deb
```

and got:
```
--2016-05-03 11:40:59--  https://grafanarel.s3.amazonaws.com/builds/grafana_3.0.0-beta71462173753_amd64.deb
Resolving grafanarel.s3.amazonaws.com (grafanarel.s3.amazonaws.com)... 54.231.12.241
Connecting to grafanarel.s3.amazonaws.com (grafanarel.s3.amazonaws.com)|54.231.12.241|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 40509924 (39M) [application/x-debian-package]
Saving to: 'grafana_3.0.0-beta71462173753_amd64.deb'

grafana_3.0.0-beta7  63%[===========>        ]  24.42M   340KB/s    in 78s

2016-05-03 11:42:17 (320 KB/s) - Read error at byte 25606798/40509924 (error:1408F119:SSL routines:SSL3_GET_RECORD:decryption failed or bad record mac). Retrying.

--2016-05-03 11:42:18--  (try: 2)  https://grafanarel.s3.amazonaws.com/builds/grafana_3.0.0-beta71462173753_amd64.deb
Connecting to grafanarel.s3.amazonaws.com (grafanarel.s3.amazonaws.com)|54.231.12.241|:443... connected.
HTTP request sent, awaiting response... 206 Partial Content
Length: 40509924 (39M), 14903126 (14M) remaining [application/x-debian-package]
Saving to: 'grafana_3.0.0-beta71462173753_amd64.deb'

grafana_3.0.0-beta7 100%[++++++++++++=======>]  38.63M   357KB/s    in 46s

2016-05-03 11:43:04 (319 KB/s) - 'grafana_3.0.0-beta71462173753_amd64.deb' saved [40509924/40509924]
```
The reason I'm showing this is because it's the same error I got with platformio, but wget retries to get failed packets and eventually succeeds in getting the whole file.

Not sure if something similar could be implemented in platformio?

ivankravets, 2016-05-04:
Thanks, I catch this error and force to HTTP mirror in a case with SSL errors.

closed 2016-05-04 by ivankravets

bzamecnik, 2018-05-11:
Downloading root certificates over HTTP might not be much secure way of doing that. But it seems that HTTPS certificate on http://www.cacert.org/ fails... Strange.

meftehSafa, 2019-12-04:
You can simply tap this command on the terminal "sudo pip install --upgrade pip" and everything will work very well

commit aca646e6 (2016-05-04): Use HTTP mirror for Package Manager in case with SSL errors // Resolve #645