platformio/platformio-core #646: PIO build fails when esp8266/nodemcu2 board is used

supunj, 2016-05-02:
Hi,

I've recently migrated my legacy IoT(https://github.com/supunj/libesp8266) project to PIO(https://github.com/supunj/libesp8266pio). All went well except for one which I believe 
may be related to PIO pulling a different version of esp8266 SDK.

Below is the error I get

In file included from .pioenvs/nodemcuv2/ESP8266WiFi/WiFiClient.h:25:0,
from .pioenvs/nodemcuv2/ESP8266WiFi/ESP8266WiFi.h:39,
from .pioenvs/nodemcuv2/ESP8266mDNS/ESP8266mDNS.h:46,
from src/esp8266util.cpp:6:
/home/travis/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
.pioenvs/nodemcuv2/FrameworkArduino/Arduino.h:254:18: error: expected unqualified-id before '(' token 

You should be able to see the same in the travis CI log.
[log.txt](https://github.com/platformio/platformio/files/245113/log.txt)

This project runs on esp8266 nodemcu2.

When I dug little deeper I figured that Arduino.h had something it didnt have in the legacy SDK.

//#ifndef _GLIBCXX_VECTOR
// arduino is not compatible with std::vector
//#define min(a,b) ((a)<(b)?(a):(b))
//#define max(a,b) ((a)>(b)?(a):(b))
//#endif

Once I commented out the above lines, it worked fine.

I'm not sure if I'm doing something wrong but I do use std::vector in the project.

Let me know if you need further details.

ivankravets, 2016-05-02:
@igrr do you have any ideas?

igrr, 2016-05-02:
Sure! Including STL headers from the same file which includes "Arduino.h" pretty much doesn't work.

`Arduino.h` has to provide `min` and `max` as defines for compatibility reasons, but alas, that causes issues with STL because there `min` and `max` are function names.
There is a ticket filed under our repository which covers this issue, so i think it's out of scope for platformio and may be closed.

Workaround: add `#undef min` and `#undef max` after `#include <Arduino.h>` before you include STL headers.

supunj, 2016-05-02:
Thanks @igrr and @ivankravets !
 that was really quick. For now I'll stick to the workaround and mark this as closed as it is reported elsewhere.

closed 2016-05-02 by supunj

mithatolut, 2018-07-23:
had to use 

#undef min
#undef max
#include <vector>

to get the compatible min  and max ...