platformio/platformio-core #647: Update intel_arc32 with WiFi101 fixes.

nrh, 2016-05-04:
- [X] Bug report. If you’ve found a bug, please provide information below.

### Configuration

**Darwin jerkcity.local 15.5.0 Darwin Kernel Version 15.5.0: Mon Apr  4 19:07:49 PDT 2016; root:xnu-3248.50.20~6/RELEASE_X86_64 x86_64**:

**PlatformIO, version 2.9.1**

### Description of problem

WiFi101 (https://github.com/arduino-libraries/WiFi101) won't currently build on platformio due to, I think, missing updates from recent `framework-arduinointel` / https://github.com/01org/corelibs-arduino101, the fixes are described here: https://github.com/arduino-libraries/WiFi101/issues/51

#### Steps to Reproduce
```
$ mkdir tmp; cd tmp
$ platformio init -b genuino101
$ cd lib; git clone git@github.com:arduino-libraries/WiFi101
$ cd ..; cp lib/WiFi101/examples/ScanNetworks/ScanNetworks.ino src/main.ino
$ platformio run
```
### Actual Results

https://gist.github.com/nrh/d284a1b765cef2705f2ef52bb864e60f

valeros, 2016-05-05:
Hi @nrh 
Thanks for the hint, but we are trying to use a stable version of the framework. 
You can easily update `arc32` core by yourself, just replace old one here: `/home/user/.platformio/packages/framework-arduinointel`

nrh, 2016-05-05:
👍

closed 2016-05-05 by nrh