platformio/platformio-core #650: DigitalIO library support for Bean Loader

MMIX2009, 2016-05-06:
I am working on a Bean project that requires the use of analog signal inputs and I2C devices. As I understand it, the Bean (https://punchthrough.com/bean) has the same two pins assigned to ADC and I2C functions. 

Do you have any plans to support software I2C libraries such as DigitalIO http://platformio.org/lib/show/200/DigitalIO or SoftI2CMaster https://github.com/todbot/SoftI2CMaster for the Bean Loader?

Thanks,

ivankravets, 2016-05-06:
@mplewis please answer.

MMIX2009, 2016-05-09:
Just checking to see if you had a chance to consider this. 
I've posted the question on Beantalk too.
http://beantalk.punchthrough.com/t/cloud-compiler-library-requests/1101/86?u=mmix2009

Thanks,

ivankravets, 2016-05-10:
Thanks. I close this issue because it isn't related to PlatformIO Repository.

closed 2016-05-10 by ivankravets

mplewis, 2016-05-13:
Thanks for posting @MMIX2009! That thread on Beantalk is the best way to reach us for Cloud Compiler library requests.

MMIX2009, 2016-05-18:
Hi @mplewis,

Thanks for responding. Any chance you can support this library in the Bean Cloud Compiler soon? I have an active project where I need this and using the Arduino IDE + Windows loader has not worked for me. Thanks.

MMIX2009, 2016-05-18:
To your point: I am hoping for a response in the Beantalk thread. Thanks.