platformio/platformio-core #651: Large Arduino Micro project built with PIO crashes, works when built with Arduino IDE

digitalstreamio, 2016-05-07:
# Configuration:

Board: RedBearLabs Blend Micro board (Arduino Micro + NRF8001) 
PlatformIO Version: 2.9.1
Operating System: Fedora 23
Framework: Arduino
Libraries: BLEPeripheral

# Issue: 

When the program is built with pio and uploaded to the board, it crashes when accessing Serial. Same source code built with Arduino IDE works fine.

# Workaround

The only way to work around the linker issue below was to embed BLEPeripheral library directly in the project/src (not project/lib) directory instead of referencing it in platformio.ini. This way the build process skips part of the static library linking step which seems to produce firmware that crashes on arduino micro.

# Details:

The issue seems to be related to the way firmware is linked. 

* pio compiled object files linked using pio generated command (outcome: firmware crashes)

/home/user/.platformio/packages/toolchain-atmelavr/bin/avr-g++ -o .pioenvs/blendmicro8/firmware.elf -Os -mmcu=atmega32u4 -Wl,--gc-sections,--relax .pioenvs/blendmicro8/src/main.o -L/home/sebby/.platformio/packages/ldscripts -L.pioenvs/blendmicro8 -Wl,--start-group .pioenvs/blendmicro8/libFrameworkArduinoVariant.a .pioenvs/blendmicro8/libFrameworkArduino.a -lm .pioenvs/blendmicro8/libSPI.a .pioenvs/blendmicro8/libBLEPeripheral_ID259.a -Wl,--end-group

* pio compiled object files linked using Arduino IDE method (outcome: firmware no longer crashes)

/home/user/.platformio/packages/toolchain-atmelavr/bin/avr-g++ -w -Os -Wl,--gc-sections -mmcu=atmega32u4  -o .pioenvs/blendmicro8/firmware.elf ".pioenvs/blendmicro8/SPI/SPI.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEAttribute.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEBondStore.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLECentral.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLECharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEConstantCharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEDescriptor.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEDevice.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEFixedLengthCharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEHID.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEHIDPeripheral.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEHIDReportMapCharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEHIDReportReferenceDescriptor.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEKeyboard.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLELocalAttribute.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEMouse.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEMultimedia.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEPeripheral.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEProgmemConstantCharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLERemoteAttribute.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLERemoteCharacteristic.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLERemoteService.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEService.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLESystemControl.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLETypedCharacteristics.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEUtil.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/BLEUuid.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/EddystoneBeacon.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/iBeacon.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/nRF51822.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/nRF8001.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/utility/aci_queue.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/utility/acilib.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/utility/hal_aci_tl.o" ".pioenvs/blendmicro8/BLEPeripheral_ID259/utility/lib_aci.o" ".pioenvs/blendmicro8/src/main.o" ".pioenvs/blendmicro8/libFrameworkArduino.a" "-L.pioenvs/blendmicro8" -lm

# Source Code:

* platformio.ini

[env:blendmicro8]
platform = atmelavr
framework = arduino
board = blendmicro8
lib_install = 259
build_flags = -DARDUINO_BLEND_MICRO_8MHZ

* main.cpp

[main.cpp.txt](https://github.com/platformio/platformio/files/253631/main.cpp.txt)

digitalstreamio, 2016-05-07:
Changing optimization options from -Os to -O in CCFLAGS and LINKFLAGS does not change the behavior observed.

valeros, 2016-05-10:
Hi @digitalstreamio, thanks for the report!
For now, I can't say what is wrong, but looks like `blePeripheral` and `Serial` objects are adjacently linked and for some reason `memset` function from `blePeripheral` constructor writes an excessive byte to `Serial` memory section.

Could you please try next [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script) with your board?
`extra_script.py`:
```
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    LIBS=[env['LIBS'][::-1]]
)
```
`platformio.ini`:
```
[env:blendmicro8]
platform = atmelavr
framework = arduino
board = blendmicro8
lib_install = 259
build_flags = -DARDUINO_BLEND_MICRO_8MHZ
extra_script = extra_script.py
```

digitalstreamio, 2016-05-10:
@valeros Nice work, the script does seem to address the issue. I can confirm that with the extra_script the firmware works properly and without it crashes as before. Thanks!

ivankravets, 2016-09-14:
@digitalstreamio could you reproduce this issue with new PlatformIO 3.0?

ivankravets, 2016-11-27:
@digitalstreamio please open new issue https://github.com/platformio/platform-atmelavr/issues if you have problem with PlatformIO 3.1

closed 2016-11-27 by ivankravets