platformio/platformio-core #652: Using `src_filter` with only one item in the list of argument causes a fatal build error on Windows

sarfata, 2016-05-08:
- [X ] Bug report. If you’ve found a bug, please provide information below.

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 2.9.1

### Description of problem

Using `src_filter` with only one item in the list of argument causes a fatal build error on Windows.

#### Steps to Reproduce

1. Create a new project with platform teensy, framework arduino, board teensy31.
2. Create a sub-folder `folder1` with one simple `main.cpp` in it
3. Edit the `platformio.ini` file and add `src_filter = +<folder1/>`

### Actual Results

````
C:\Users\thomas\Documents\GitHub\test> platformio run
[05/08/16 20:18:44] Processing mybaseenv (platform: teensy, src_filter: +<folder1/>, f
ramework: arduino, board: teensy31)
-------------------------------------------------------------------------------------
The system cannot find the file specified.
============================ [ ERROR ] Took 0.03 seconds ============================
````

With `-v` the results are slightly different:
````
C:\Users\thomas\Documents\GitHub\test> platformio run -v
[05/08/16 20:18:49] Processing mybaseenv (platform: teensy, src_filter: +<folder1/>, f
ramework: arduino, board: teensy31)
-------------------------------------------------------------------------------------
.
============================ [ ERROR ] Took 0.03 seconds ============================
````
### Expected Results

Project should build without error on Windows, [as it does on Travis](https://travis-ci.org/sarfata/platformio-bug-srcfilter).

### If problems with PlatformIO Build System:

See also full project here: https://github.com/sarfata/platformio-bug-srcfilter

**The content of `platformio.ini`:**
```ini
[env:mybaseenv]
platform = teensy
framework = arduino
board = teensy31
# will not work with platformio 2.9.1
src_filter = +<folder1/>
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("looping...");
  delay(1000);
}
```

### Additional info

 - Just adding some text at the end of the line (for example a comment or a folder that does not exist) is enough to workaround the issue.
 - Tested under Linux and the error does not happen there. See [travis build report](https://travis-ci.org/sarfata/platformio-bug-srcfilter).
 - Tested in the command line and in Atom with same results.

ivankravets, 2016-05-08:
Thanks a lot for the detailed report! 👍 

The bug is linked with SCons command parser. I'll think how to fix it.

closed 2016-05-10 by ivankravets

sarfata, 2016-05-11:
:+1: Thanks for the quick fix!

commit 056cc08d (2016-05-10): Fix issue with ``src_filter`` option for Windows OS // Resolve #652