platformio/platformio-core #653: Including <algorithm> breaks the build

corani, 2016-05-09:
### Configuration

**Operating system**:
Ubuntu Linux 14.04 LTS (64 bit)

**PlatformIO Version** (`platformio --version`):
2.9.0

### Description of problem
Since a recent update, simply including &lt;algorithm&gt; suddenly breaks compilation badly. 

#### Steps to Reproduce

1. Take a C++ project that used to work fine (build_flags = -lstdc++ -lsupc++)
2. Include &lt;algorithm&gt; anywhere

### Actual Results
```
In file included from ~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:60:0,
from ~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
from src/main.cpp:7:
~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/algorithmfwd.h:360:41: error: macro "max" passed 3 arguments, but takes just 2
max(const _Tp&, const _Tp&, _Compare);
^
~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/algorithmfwd.h:371:41: error: macro "min" passed 3 arguments, but takes just 2
min(const _Tp&, const _Tp&, _Compare);
^
~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/algorithmfwd.h:394:30: error: macro "min" requires 2 arguments, but only 1 given
min(initializer_list<_Tp>);
^
~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/algorithmfwd.h:402:30: error: macro "max" requires 2 arguments, but only 1 given
max(initializer_list<_Tp>);
```
And a 1000+ further lines of errors

### Expected Results

ivankravets, 2016-05-09:
This is not a bug of PlatformIO. The issue is linked with the ESP8266 for Arduino. See https://github.com/platformio/platformio/issues/646#issuecomment-216244875

closed 2016-05-09 by ivankravets

corani, 2016-05-09:
Thanks for the pointer. A liberal sprinkling of `undef`s worked around the problem for me, but I'd like to stress again that this code worked just fine the last time I compiled it (maybe a week or two ago).

ivankravets, 2016-05-09:
We updated ESP8266 for Arduino package 1 week ago to the latest version 2.2.0 (by user requests). See https://github.com/esp8266/Arduino/releases . Looks like you used 2.1.0 before.

Please sorry for the inconveniences. We are working on the PlatformIO 3.0 and #479 issue where you will be able to fix the version of `platform` or even fork/create own platform in "a few clicks".