platformio/platformio-core #654: Bug. Compiling TFT_ILI9163C-Pre-Release-1.0r6

St007, 2016-05-11:
Tried to use that library ( https://github.com/sumotoy/TFT_ILI9163C ) and compilation fails with Platformio, but it works with latest Arduino IDE.
**.pioenvs\esp12e\TFT_ILI9163C\_fonts\arial_x2.c:1582:1: error: unknown type name 'tFont'**:

**The content of `platformio.ini`:**
```ini
[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
```

**arial_x2.c**
```cpp
#include <stdint.h>

static const uint8_t image_data_arial_x2_0x20[1] 
#if defined(_FORCE_PROGMEM__) 
PROGMEM 
#endif  
= { 0x00 };////character: ' '

static const tImage arial_x2_0x20 
#if defined(_FORCE_PROGMEM__) 
PROGMEM 
#endif 
= { image_data_arial_x2_0x20,
    2, 1};//character: ' ' , height: 9, (charW,total bytes)
...
const tFont arial_x2 = { 96, arial_x2_array,0,9,0 };
```
**TFT_ILI9163C_fontDescription.h**
```cpp
#ifndef __TFT_ILI9163CFONTD_H
	#define __TFT_ILI9163CFONTD_H
	#include <stdio.h>
	
	#if defined(_FORCE_PROGMEM__)
		typedef struct PROGMEM {//__PRGMTAG_
			const uint8_t 	*data;
			uint8_t 		image_width;
			int				image_datalen;
		} tImage;
	
		typedef struct {
			uint8_t 		char_code;
			const tImage 	*image;
		} tChar;
	#else
		typedef struct {
			const uint8_t 	*data;
			uint8_t 		image_width;
			int				image_datalen;
		} tImage;
	
		typedef struct {
			uint8_t 		char_code;
			const tImage 	*image;
		} tChar;
	#endif

	typedef struct {
			uint8_t 		length;
			const tChar 	*chars;
			uint8_t			font_width;
			uint8_t			font_height;
			bool 			rle;
	} tFont;
	
#endif
```

**TFT_ILI9163C_fontDescription.h**
```cpp
ifndef _TFT_ILI9163CLIB_H_
#define _TFT_ILI9163CLIB_H_

#include "Arduino.h"
#include "Print.h"

#include <limits.h>
#include "pins_arduino.h"
#include "wiring_private.h"
#include <stdio.h>
#include <stdlib.h>
#include <SPI.h>

#include "_settings/TFT_ILI9163C_settings.h"
#include "_includes/TFT_ILI9163C_cpuCommons.h"
#include "_settings/TFT_ILI9163C_colors.h"
#include "_includes/TFT_ILI9163C_registers.h"

#include "_includes/TFT_ILI9163C_fontDescription.h"
#include "_fonts/arial_x2.c"

#if defined(ESP8266) && !defined(_ESP8266_STANDARDMODE)
	#include <eagle_soc.h>
#endif

#if defined(SPI_HAS_TRANSACTION)
	static SPISettings ILI9163C_SPI;
#endif

```
### Additional info

valeros, 2016-05-12:
Hi @St007 !
The main problem is that PlatformIO build system differs from Arduino IDE and, moreover, this library has an unusual structure with using `.c` file as a header.
Could you please try to rename file `arial_x2.c` into `arial_x2.h` and fix include in `TFT_ILI9163C.h:85`
```cpp
#include "_fonts/arial_x2.h"
```
Thanks!

St007, 2016-05-17:
Thank you very much for clearance, you suggestion solves the issue.

closed 2016-05-17 by St007

ivankravets, 2016-05-18:
Please grant PlatformIO repository with star ⭐ https://github.com/platformio/platformio/stargazers

Thanks a lot!