platformio/platformio-core #660: Missing deb-package dependency: python-requests

dothebart, 2016-05-16:
- [x ] Bug report. If you’ve found a bug, please provide information below.
Downloaded & installed the .deb file; platformio wouldn't start because of the module `request` missing.

    apt-get install python-request 

fixed the issue.
So, the .deb should depend on this package.

ivankravets, 2016-05-16:
How did you install `platformio`?

dothebart, 2016-05-16:
I downloaded the .deb via the download link, and `dpkg -i platformio-atom-linux-x86_64.deb`ed it.

ivankravets, 2016-05-16:
Hm... In this case, PlatformIO IDE installs own version of PlatformIO CLI and creates special Python environment.

Do you have PlatformIO CLI globally installed in your system?

dothebart, 2016-05-16:
yes, installed the .deb as root. Its living in /usr/local.

ivankravets, 2016-06-15:
Please provide me an output of `which pio`. Thanks.

ivankravets, 2016-07-05:
Please reopen if you still need help

closed 2016-07-05 by ivankravets

dothebart, 2016-07-05:
sorry, I didn't need help. I fixed it for myself by installing the missing dependency. This issue was about you fixing it for other users that may run into this trap.

ivankravets, 2016-07-05:
@dothebart Thanks a lot for the report!