platformio/platformio-core #662: Visual Studio error parsing support

EraYaN, 2016-05-18:
What kind of issue is this?
- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
2.9.1

### Description of problem
Visual Studio displays the error in the build log, but doesn't parse them.

#### Steps to Reproduce

1. Build project in visual with build errors
2. Check Errors tab

### Actual Results
Errors only contains "exited with code 1"

### Expected Results
Error has a list of all errors and warnings (clickable)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
int main(){
    -notAReaLCFILEERROR!
}
```

### Additional info
The thing needed for this to work, is something like this: http://www.codeproject.com/Articles/370890/GCCFilter-A-script-for-compiling-with-GCC-in-Visua (But please don't use perl like that, powershell might do, or some python) I'm pretty good with python but no idea where this would need to be added.

Basically a wrapper so the output is transformed so people can click their errors.

Also add the folders with the framework to the include path of the project. This way the full IntelliSense works. (People can do this manually) This can be done by the init script. It knows the board, it can install the framework folders and add those to the project.
Mine look like this:
```
.\.pioenvs\lpc1768\FrameworkMbedInc-250686161\;.\.pioenvs\lpc1768\FrameworkMbedInc549023428\;.\.pioenvs\lpc1768\FrameworkMbedInc-550770547\;.\.pioenvs\lpc1768\FrameworkMbedInc1648309630\;$(IncludePath)
```

ivankravets, 2016-05-18:
See [Known issues: IntelliSense Errors](http://docs.platformio.org/en/latest/ide/visualstudio.html#intellisense-errors).

Microsoft Visual Studio doesn't allow to cutomize IntelliSense with own toolchains.

EraYaN, 2016-05-18:
It does allow you to add the IncludePath and Source Directories. (In the project file)

This gives an almost complete IntelliSense experience. I know it's easy mode, but it works. It's not about the errors, it's about code completion and function prototypes. (Like [this](http://i.imgur.com/0SkO0oS.png) And that does add a lot to the experience. It's auto complete and refactoring is one of the reasons I use the IDE.