platformio/platformio-core #663: Cannot import libraries

djolez, 2016-05-18:
### Configuration

**Operating system**: Raspbian Jessie

**PlatformIO Version** (`platformio --version`): 2.9.1

### Description of problem
Hi, I tried to include OneWire (and also DHT11) library using library manager but I get `fatal error: OneWire.h: No such file or directory`, I checked in the `~/.platformio/lib` folder and the the library is there.. I saw that there were similar problems in some older versions but as far as I know it was fixed. Thanks in advance

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno
```

ivankravets, 2016-05-18:
Please provide
1. `ls -la ~/.platformio/lib`
2. Source code of the project (`src` directory)

djolez, 2016-05-18:
`ls-la ~/.platformio/lib DHT11_ID55/  OneWire_ID1`

Sorry for bad formatting and the way I attach the source code (I have only that file), I didn't use Github before.

[main.txt](https://github.com/platformio/platformio/files/271106/main.txt)

ivankravets, 2016-05-18:
What is it `#include<OneWire.h>`? You need a space between `#include` and `<..>`.

djolez, 2016-05-18:
Please sorry, I have problems with copying text since I connect via SSH from Windows to Raspberry, it must have lost formatting at some point because I was copying that part, now it uploads the code.. Sorry again for bothering you

ivankravets, 2016-05-18:
Please grant PlatformIO repository with star ⭐ https://github.com/platformio/platformio/stargazers

Thanks a lot!

closed 2016-05-18 by ivankravets