platformio/platformio-core #664: .pioenvs caching breaks strange behavior on 2nd run

nargetdev, 2016-05-19:
------------------------------------------------------------------
### Description of problem

The first time I run  **$ pio run** my project builds just fine and generates appropriate things in .pioenvs folder. The second time I **$ pio run** new things will be generated in .pioenvs which cause many compile errors.

Specifically, on the second run, it appears to be pulling in an entirely unrelated library (one built for a different platform entirely).

This is easy enough to cope with as follows:  **$rm -rf .pioenvs && pio run**
effectively causing every run to be the "first" run and allowing me to work as normal.

Anybody else seen this?

### Configuration

**Operating system**: OS X

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.9.1

#### Steps to Reproduce
1.  Import Arduino project with board Due (using Arduino libraries folder)
2.  run $pio run
3.  run $pio run again

### Actual Results
.pioenvs populates with "grbl" library on second run

### Expected Results
.pioenvs stays the same after first being generated

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:due]
platform=atmelsam
framework=arduino
board=due

[platformio]
src_dir=.
lib_dir=~/Documents/Arduino/Libraries```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

 // Simple demo of three threads
// LED blink thread, print thread, and idle loop
#include <FreeRTOS_ARM.h>
// Redefine AVR Flash string macro as nop for ARM
#undef F
#define F(str) str

const uint8_t LED_PIN = 13;

volatile uint8_t count = 0;

// handle for blink task
TaskHandle_t blink;

//------------------------------------------------------------------------------
// high priority for blinking LED
static void vLEDFlashTask(void *pvParameters) {
  pinMode(LED_PIN, OUTPUT);

  // Flash led every 200 ms.
  for (;;) {
    // Turn LED on.
    digitalWrite(LED_PIN, HIGH);

    // Sleep for 50 milliseconds.
    vTaskDelay((50L * configTICK_RATE_HZ) / 1000L);

    // Turn LED off.
    digitalWrite(LED_PIN, LOW);

    // Sleep for 150 milliseconds.
    vTaskDelay((150L * configTICK_RATE_HZ) / 1000L);
  }
}
//------------------------------------------------------------------------------
static void vPrintTask(void *pvParameters) {
  while (1) {
    // Sleep for one second.
    vTaskDelay(configTICK_RATE_HZ);

    // Print count for previous second.
    Serial.print(F("Count: "));
    Serial.print(count);

    // Print unused stack for threads.
    Serial.print(F(", Unused Stack: "));
    Serial.print(uxTaskGetStackHighWaterMark(blink));
    Serial.print(' ');
    Serial.print(uxTaskGetStackHighWaterMark(0));
    Serial.print(' ');
    Serial.println(uxTaskGetStackHighWaterMark(xTaskGetIdleTaskHandle()));

    // Zero count.
    // count = 0;
  }
}
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(9600);
  // wait for Leonardo
  while(!Serial) {}

  // create blink task
  xTaskCreate(vLEDFlashTask,
    "Task1",
    configMINIMAL_STACK_SIZE + 50,
    NULL,
    tskIDLE_PRIORITY + 2,
    &blink);

  // create print task
  xTaskCreate(vPrintTask,
    "Task2",
    configMINIMAL_STACK_SIZE + 100,
    NULL,
    tskIDLE_PRIORITY + 1,
    NULL);

  // start FreeRTOS
  vTaskStartScheduler();

  // should never return
  Serial.println(F("Die"));
  while(1);
}
//------------------------------------------------------------------------------
// WARNING idle loop has a very small stack (configMINIMAL_STACK_SIZE)
// loop must never block
void loop() {
  while(1) {
    // must insure increment is atomic
    // in case of context switch for print
    noInterrupts();
    count++;
    interrupts();
  }
}
```

### Additional info

nargetdev, 2016-05-19:
Hello, I have an appropriate workaround for the time being.  This could be scripted easily and does not break Arduino compatibility.

Firstly:
Remove **lib_dir=~/Documents/Arduino/Libraries** from platformio.ini

The script:
Regex lib includes of the following pattern.. **#include <FreeRTOS_ARM.h>**
Find those folders in **~/Documents/Arduino/Libraries** and copy them to **lib** ONLY INCLUDING **src** subdir.

Shall I script this, and add the appropriate option in the Atom IDE "import Arduino IDE project" module?

ivankravets, 2016-05-20:
@valeros please add these 3 libs to PlatformIO Library Registry https://github.com/greiman/FreeRTOS-Arduino/tree/master/libraries

valeros, 2016-05-25:
These libs are in our library manager now:
http://platformio.org/lib/show/324/FreeRTOS_ARM
http://platformio.org/lib/show/323/FreeRTOS_AVR
http://platformio.org/lib/show/322/SdFat

ivankravets, 2016-05-28:
Please reopen if you still need a help

closed 2016-05-28 by ivankravets