platformio/platformio-core #665: STM32F0 upload failing

zgoda, 2016-05-20:
```
$ pio run --target upload
[Fri May 20 12:05:58 2016] Processing disco_f051r8 (platform: ststm32, board: disco_f051r8, framework: mbed)
------------------------------------------------------------------------------------------------------------------------
MethodWrapper(["upload"], [".pioenvs/disco_f051r8/firmware.bin"])
scons: *** [upload] Explicit exit, status Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms this can be a USB flash drive (i.e. /media/<user>/<device name>)

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms this can be a USB flash drive (i.e. /media/<user>/<device name>)
```

Port is /dev/stlinkv2_6 so let's follow advice in another approach:

```
$ pio run --target upload --upload-port /dev/stlinkv2_6 
.clang_complete  .gitignore       .pioenvs/        src/             
.gcc-flags.json  lib/             platformio.ini   .travis.yml      
jazg@JAZG-LENOVO:~/work/stm32_test $ pio run --target upload --upload-port /dev/stlinkv2_6 
[Fri May 20 12:06:22 2016] Processing disco_f051r8 (platform: ststm32, board: disco_f051r8, framework: mbed)
------------------------------------------------------------------------------------------------------------------------
MethodWrapper(["upload"], [".pioenvs/disco_f051r8/firmware.bin"])
scons: *** [upload] /dev/stlinkv2_6/firmware.bin: Not a directory
```

Port is accessible, of course:

```
$ ls -la /dev/stlinkv2_6 
lrwxrwxrwx 1 root root 15 maj 20 11:53 /dev/stlinkv2_6 -> bus/usb/003/003
```

Code builds without problems:

```
$ pio run
[Fri May 20 12:13:34 2016] Processing disco_f051r8 (platform: ststm32, board: disco_f051r8, framework: mbed)
------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/disco_f051r8/src/main.o 

[ ... skipped for brevity ... ]

"arm-none-eabi-size" -B -d .pioenvs/disco_f051r8/firmware.elf
arm-none-eabi-objcopy -O binary .pioenvs/disco_f051r8/firmware.elf .pioenvs/disco_f051r8/firmware.bin
text	   data	    bss	    dec	    hex	filename
12988	    124	    548	  13660	   355c	.pioenvs/disco_f051r8/firmware.elf
============================================= [SUCCESS] Took 0.91 seconds =============================================
```

ivankravets, 2016-05-20:
Could you copy manually without root permissions? 
```
cp .pioenvs/disco_f051r8/firmware.elf /dev/stlinkv2_6
```

zgoda, 2016-05-20:
```
$ LANG=C cp .pioenvs/disco_f051r8/firmware.elf /dev/stlinkv2_6
cp: error writing '/dev/stlinkv2_6': Invalid argument
```

ivankravets, 2016-05-20:
Do you need any help?

zgoda, 2016-05-20:
I'd like to know where is the problem - my board, my environment or PlatformIO components. I can replace board if it's not supported or I can try to fix my environment.

This is [this exact kit](https://www.element14.com/community/docs/DOC-67559/l/discovery-kit-for-stm32f0-with-st-linkv2)

zgoda, 2016-05-20:
Texane stlink tool works fine:

```
$ st-flash write .pioenvs/disco_f051r8/firmware.bin 0x8000000
2016-05-20T12:46:00 INFO src/common.c: Loading device parameters....
2016-05-20T12:46:00 INFO src/common.c: Device connected is: F0 device, id 0x20006440
2016-05-20T12:46:00 INFO src/common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2016-05-20T12:46:00 INFO src/common.c: Attempting to write 13112 (0x3338) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003000 erased
2016-05-20T12:46:00 INFO src/common.c: Finished erasing 13 pages of 1024 (0x400) bytes
2016-05-20T12:46:00 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2016-05-20T12:46:00 INFO src/common.c: Successfully loaded flash loader in sram
 12/12 pages written
2016-05-20T12:46:01 INFO src/common.c: Starting verification of write complete
2016-05-20T12:46:01 INFO src/common.c: Flash written and verified! jolly good!
```

valeros, 2016-05-20:
Hi @zgoda 
Looks like your kit only supports uploading over `STlink`.
Could you try `upload_protocol = stlink` option in your `platformio.ini`?

zgoda, 2016-05-20:
Thank you @valeros, this works fine.

ivankravets, 2016-05-20:
@zgoda Please give PlatformIO repository a star ⭐ https://github.com/platformio/platformio/stargazers

Thanks a lot!

ivankravets, 2016-05-20:
@valeros please specify `stlink` upload protocol for the boards without ARM mbed bootloader.

closed 2016-06-02 by ivankravets

bfarayev, 2016-06-28:
Thanks @valeros Adding `upload_protocol = stlink` to `platform.ini` solved the issue

ivankravets, 2016-06-28:
@bfarayev it should work automatically since 2.9.2 release. What is your PlatformIO CLI version? Could you share here your `platformio.ini`?

bfarayev, 2016-06-28:
@ivankravets well, I'm using 2.10.3. My `platform.ini` file was like this:

```
[env:nucleo_f411re]
platform = ststm32
framework = mbed
board = nucleo_f411re
```

I was suspecting if it's because I created this project long before. However, I initialized a new one and again, `platform.ini` file looks similar:

```
[env:disco_f407vg]
platform = ststm32
framework = mbed
board = disco_f407vg
```

ivankravets, 2016-06-28:
@bfarayev Thanks, I got you now. The `stlink` protocol is set by default only for the `disco_*` boards where the `mbed disk` is used for `nucleo_*`.

In this case, you need manually force the `upload_protocol` as you do it.

TravisJoe, 2017-05-03:
This should probably be done on the Nucleo boards as well. Just ran into the same issue.

Goldenstarc, 2018-11-16:
> Texane stlink tool works fine:
>
> ```
> $ st-flash write .pioenvs/disco_f051r8/firmware.bin 0x8000000
> 2016-05-20T12:46:00 INFO src/common.c: Loading device parameters....
> 2016-05-20T12:46:00 INFO [...]

can you say that whats exactly in you ini file ?
I hace to save firmware.bin somewhere and flash it using st-flash in another terminal

commit 03d9351d (2016-06-02): Add "stlink" as the default uploader for STM32 Discovery boards // Resolve #665
commit 06e42dc6 (2016-06-02): Merge branch 'develop' into feature/platformio-30

* develop:
  Version bump to 2.9.2 (issues #641, #645, #648, #652, #664, #665, #666, #671, #674)
  Fix issue with ARM mbed framework when abstract class breaks compile for LPC1768 // Resolve #666
  Fix issue with ARM mbed framework and multiple definition errors   on FRDM-KL46Z board // Resolve #641
  List embedded boards in docs
  Fix multiple definition in mbed framework when using abstract class // Issue #641, #666
  Add "stlink" as the default uploader for STM32 Discovery boards // Resolve #665
  Implement grep serial ports for Windows
  R [... cut]