platformio/platformio-core #669: Upload to target eventually fails after multiple uploads due to "no space left on device"

gilandose, 2016-05-24:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:  Linux

**PlatformIO Version** (`platformio --version`):PlatformIO, version 2.9.1

### Description of problem
`

[Tue May 24 16:13:58 2016] Processing nucleo_f401re (platform: ststm32, upload_port: /media/rgilmore/NODE_F401RE, board: nucleo_f401re, framework: mbed, bui
ld_flags: -lstdc++ -lsupc++)
--------------------------------------------------------------------------------
MethodWrapper(["upload"], [".pioenvs/nucleo_f401re/firmware.bin"])
scons: *** [upload] No space left on device
========================= [ ERROR ] Took 1.13 seconds =========================`

#### Steps to Reproduce

1.Upload a design to device multiple times.
2.After a certain number of times error appears
3.unplug and replug in device to fix

### Actual Results
Seems to keep incrementing write area until no space left

### Expected Results
Should erase flash and upload new program.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```

[env:nucleo_f401re]
platform = ststm32
framework = mbed
board = nucleo_f401re
build_flags = -lstdc++ -lsupc++
upload_port=/media/rgilmore/NODE_F401RE
```

valeros, 2016-06-02:
Hi @gilandose, sorry for the late reply! Our uploader just copies a firmware to `mbed` media disk and nothing more, so probably the problem isn't in PlatformIO.
As I know `nucleo_f401re` has the embedded `st-link` programmer on the board. Have you considered using `tool-stlink` as an alternative uploader?
```
[env:nucleo_f401re]
platform = ststm32
framework = mbed
board = nucleo_f401re
upload_protocol = stlink
```

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-ststm32/issues/23

closed 2016-11-27 by ivankravets