platformio/platformio-core #670: Feature Request - It would be nice to have the ability to copy all output from the serial monitor to a text file, or even have an automatic logging option for the serial monitor.

rtrevorclark, 2016-05-26:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

bhavyangupta, 2016-08-25:
If you are using linux then you can use the following to write the output to a file:

```
platformio serialports monitor --port=<port> --baud=<baud> >> output_file
```

s4m0r4m4, 2017-02-04:
Good idea!  this would be a great feature.

ivankravets, 2019-09-04:
Sorry for the delay. I'm going to close this issue because we don't plan to fix it soon. We use PySerial or CLI where you can forward buffer into file. See https://github.com/platformio/platformio-core/issues/670#issuecomment-242491591

For more advanced SerialMonitor, please take a look at https://freeware.the-meiers.org/

closed 2020-03-17 by ivankravets

ivankravets, 2020-03-17:
Please upgrade to the latest version via `pio upgrade --dev` and re-test with `log2file` filter. See docs https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html

Does it work?

commit 0df72411 (2020-03-17): Device Monitor Filter API, implement  "time" and "log2file" filters // Resolve #981 Resolve #670