platformio/platformio-core #672: #includes in included files are not recognized

jadedanemone, 2016-05-28:
### Configuration

**Operating system**: Mac 10.11.4

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.9.1

### Description of problem
Certain `#include`s in `#include`s are not recognized.

#### Steps to Reproduce

1. With the board set to Adafruit HUZZAH ESP8266
2. Duplicate the "BasicOTA.ino" example from https://github.com/esp8266/Arduino/tree/master/libraries/ArduinoOTA/examples/BasicOTA
3. Remove the `#include <ESP8266mDNS.h>` from BasicOTA.ino.

### Actual Results
```
.pioenvs/huzzah/ArduinoOTA/ArduinoOTA.cpp:21:25: fatal error: ESP8266mDNS.h: No such file or directory
#include <ESP8266mDNS.h>
^
compilation terminated.
```

### Expected Results
Successful compilation.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:huzzah]
platform = espressif
framework = arduino
board = huzzah
```

**Source file to reproduce issue:**
```cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "..........";
const char* password = "..........";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
```

### Additional info
This seems to be #189 and #195 again.
Note: I have provided a generic way to reproduce the problem, and will work around it in my current code. However, this is not the desired or intended behavior, so, bug report...

ivankravets, 2016-05-28:
Deep scanning for dependent libraries is turned off by default. See [lib_dfcyclic](http://docs.platformio.org/en/latest/projectconf.html#lib-dfcyclic).

ivankravets, 2016-05-28:
The simple way is to add `#include <ESP8266mDNS.h>` to your main source file.

jadedanemone, 2016-05-29:
So, what you are saying is "this is the intended behavior"? I'll work around that if I need to. What is the drawback to turning lib_dfcyclic on by default?

ivankravets, 2016-05-29:
> What is the drawback to turning lib_dfcyclic on by default?

We had this option enabled before by default. However, we received a few issues where that doesn't work properly. In this case, we propose developers 2 options:
1. They manually specify in source files (`src` dir) required libraries
2. They have an ability to enable deep search for dependent libraries via the special option in `platformio.ini`.

------

I'm working on PlatformIO 3.0 where Build System for libraries will be rewritten from the scratch. Maybe, I'll find better solution.

-------

closed 2016-05-30 by jadedanemone

mark-hahn, 2017-01-11:
I'm stuck on this problem.  I've tried putting `lib_ldf_mode = deep` in the ini file and putting `#include <EEPROM.h>` in every file in my project *(!)*.  I'm going to have to put the library code in my main file until this is fixed.

ivankravets, 2017-01-12:
@mark-hahn could you provide a project to reproduce this issue?

mark-hahn, 2017-01-12:
Sure,  Is there a way I can send it only to you instead of making it
public?  It's semi-private.

ivankravets, 2017-01-12:
@mark-hahn Please send to me@ikravets.com, I'll not share your project.

mark-hahn, 2017-01-12:
I reverted back to the time I had the problem and of course I can't recreate the problem.  I'm pretty sure what I reported in the OP was accurate at the time.  If I get back in that state I'll zip up the entire project folder and come back here.

Anyway, sorry for the trouble.

denzelwamburu, 2017-04-27:
Am also having the same issue.

ivankravets, 2017-04-27:
Please read http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode