platformio/platformio-core #674: Feature Request - ARM tool-openocd

marty-sullivan, 2016-05-28:
When I try to upload to my Due from Raspberry Pi I get the message "Error: The package 'tool-openocd' is not available for your system 'linux_armv7l'"

How do I either install this manually or can you guys add the package to your repo?

ivankravets, 2016-05-28:
We don't have binaries for Linux ARM. See full list https://bintray.com/platformio/dl-packages/tool-openocd/view#files

Could you provide us OpenOCD binaries?

marty-sullivan, 2016-05-28:
Sure, I could do that. Is there a specific way things need to be packaged? Or just regular build directory is fine?

ivankravets, 2016-05-28:
Take a look at these examples https://bintray.com/platformio/dl-packages/tool-openocd/view#files

marty-sullivan, 2016-05-28:
So I've built the package, but I notice that each package has a "bin/openocd.bin" but I don't know what this is or how to make it. Probably just my unfamiliarity with the platform.

ivankravets, 2016-05-28:
Please archive it and share somewhere. I'll take a look.

marty-sullivan, 2016-05-28:
Here is the package so far:

https://cornell.box.com/s/s8rflgdvo6kmnd1zdoebf5xhy12u6r6e

ivankravets, 2016-05-28:
It requires login/pass. Please attach archive here.

marty-sullivan, 2016-05-28:
sorry i forgot to edit the permissions, should work now. I'm on the road so I can't attach it atm. Thanks

ivankravets, 2016-05-28:
Thanks. I've just added this package to the repo. Please test.

marty-sullivan, 2016-05-29:
OK, looks like all is good. Thanks for the quick help.

ivankravets, 2016-05-29:
Happy Coding with PlatformIO! 😊

closed 2016-05-29 by ivankravets