platformio/platformio-core #678: Define custom boards

jerabaul29, 2016-05-31:
I did some modifications of the Due core to fix some issues I had with getting full resolution on the PWM, plus to let me choose separately the sizes of the RX and TX buffers on Serial. The core, variant and board.txt are available here:

[https://github.com/jerabaul29/ArduinoDue](https://github.com/jerabaul29/ArduinoDue)

I used it with the Arduino IDE, and it works fine (compiles, uploads, do what I want on the board).

Now I would like to use this board with Platformio (that I have as an Atom plugin). I tired to modify locally the code in the .pioenvs of a project, and to add the new core and board in /home/[username]/.platformio/packages/ but both attempts failed. In particular, I get an annoying 'fatal error: UARTClass_1.h: No such file or directory' error. While I have the UARTClass_1.h file in my core, it seems it disappears when Platformio copies it locally in one project. Any clue how I may solve this?

valeros, 2016-05-31:
Hi @jerabaul29 !
`.pioenvs` is a temporary directory, so every build will overwrite your changes.
Take a look at [creating custom board](http://docs.platformio.org/en/latest/platforms/creating_board.html) page in our docs.
Also, you will need to add your `core` and `variant` to framework package `/home/username/.platformio/packages/framework-arduinosam`
Keep in mind that your changes will be overwritten with the next framework update.

jerabaul29, 2016-05-31:
Hej!

Thanks! That is what I suspected for ```.pioenvs```, great you could confirm it was not the right place ;)

Ok, I had been through the *creating custom boards* page but too quickly, now I think I understand what I should do there. Thanks for confirming me that I should put everything in the framework-arduinosam.

Is there a way to create my own folder with cores and variants that will not be overwritten at next framework update? Like a ```framework_custom```? How to refer to it in the custom board JSON file then?

ivankravets, 2016-05-31:
> Is there a way to create my own folder with cores and variants that will not be overwritten at next framework update? Like a framework_custom? How to refer to it in the custom board JSON file then?

You will be able to do this with PlatformIO 3.0 that is still in development. If you want to try it (development version), I can explain.

jerabaul29, 2016-05-31:
Ok, great! I can wait and use the Arduino IDE in the while ;)

ivankravets, 2016-05-31:
Please follow us on [Twitter](https://twitter.com/PlatformIO_Org) and you will be notified about PlatformIO 3.0 release.

closed 2016-05-31 by ivankravets

Dubiy, 2017-03-18:
@ivankravets, can you explain how it works in 3.2? I can't find it in documentation or somewhere else. I'm trying to use custom pins_arduino.h for atmega8 instead of ~\.platformio\packages\framework-arduinoavr\variants\standard\pins_arduino.h

ivankravets, 2017-03-25:
@Dubiy 

* http://docs.platformio.org/en/latest/platforms/creating_board.html
* https://github.com/platformio/platform-atmelavr/tree/develop/boards

Dubiy, 2017-03-27:
@ivankravets, yes, but no :)
I create custom board, and it's work well. But this custom board must use custom file pins_arduino.h. 
Defaultly it use ~.platformio\packages\framework-arduinoavr\variants\standard\pins_arduino.h This file is outside the project directory, so I can't add it to repository...

ivankravets, 2017-03-27:
@Dubiy 

* http://docs.platformio.org/en/latest/projectconf.html#dynamic-build-flags
* http://docs.platformio.org/en/latest/projectconf.html#extra-script

Take a look at `Before/Pre and After/Post actions` by the last link. You can use pre-hook and copy own `variant` to base arduino framework.

If you plan to add this board to PIO, then you can make PR to https://github.com/platformio/platformio-pkg-framework-arduinoavr/tree/master/variants

yannik17, 2018-09-16:
I am confused. I cannot find the documentation of this any more. I need an easy way of creating a Arduino-based board, but with a different hardware ID and a custom pins_arduino.h file. What I've done so far is creating a myboardname.json file in my projects' boards directory and changing the hardware ID. I left build/variant unchanged, otherwise I get `<...>/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:257:26: fatal error: pins_arduino.h: No such file or directory`. For me it's unclear how to supply a custom pins_arduino.h file. Can you say me how to do it?

ivankravets, 2018-09-16:
1. Put custom “variant” to HOME_DIR/.platformio/packages/framework-arduinoavr/variants
2. Change “variant” field in board manifest

Timvrakas, 2019-04-04:
Ivan, is a Python hook still the best way to do this? I have several boards, all adafruit feather M0 at the core, but different pinouts/sercoms needed. Is there a way to include just a varients.cpp override within the project directory? No changes to the board definition .json file are generally needed.

ivankravets, 2019-04-04:
@Timvrakas yes, extra scripting is a right solution here. Also, there is a related feature request https://github.com/platformio/platformio-core/issues/1913

Timvrakas, 2019-08-18:
See https://github.com/platformio/platform-atmelsam/pull/74 for an alternative solution.

yyachim, 2019-08-23:
Is it possible to configure a project to be "board-less" with a given microcontroller (so that all of its pins are usable)?

ivankravets, 2019-08-23:
@yyachim yes, we work on that. Some part of work is already done for Atmel AVR and ST STM32 dev/platforms. So, very soon you will be able to configure the project using `bare` MCU as an identifier for `board` property.

What is your MCU?

yyachim, 2019-08-23:
@ivankravets Thanks for the quick response. I'm working on a commercial product using SAMD51.

ivankravets, 2019-08-23:
What is your SDK/API? CMSIS? Harmony? Atmel Framework?

yyachim, 2019-08-23:
I've been using the atmelsam platform with the Arduino API.

S2Doc, 2019-08-23:
@ivankravets  I am also using ATSAMD51J19 with Arduino framework.  I need this functionality for a commercial product.

ivankravets, 2019-08-23:
if you need Arduino API, then use any "board ID" which is based on your MCU and has support for Arduino.

See https://platformio.org/boards?filter%5Bmcu%5D=SAMD51J19

yyachim, 2019-08-23:
Yeah, I started with the Feather M4 board, but it doesn't provided all of the MCU pins that I need. I'm planning to copy the variant files from framework-arduinosam to my project folder and update g_APinDescription array in variant.cpp to include the missing pins. Until "board = ATSAMD51J19" is available, it sounds like this is the way to proceed?

ivankravets, 2019-08-23:
@yyachim I see. You actually need this http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files

ivankravets, 2019-08-29:
if you need only a custom `variant`, just create it in `variants` folder and override later with `board_build. variant = custom_variant`.

Timvrakas, 2019-08-30:
You mean the variants folder in the local framework package? I'm trying to sychronize development across a large group of inexperienced users, and that felt somewhat hacky, perhaps unreliable. I think not modifying the state of the installation is probably better.

You mention that a generic board option is in the works, how will that function?