platformio/platformio-core #680: Installation: Click INITIALIZE. Half-way through list of boards I own, crash.

childresss, 2016-06-02:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 7 64 bit
**PlatformIO Version** (`platformio --version`):
1.2.2, 2.9.1
### Description of problem
New/first installation. Selected boards I own (about 10 different).Clicked INITIALIZE.
After several successes, This error:
PlatformIO: Failed to initialize PlatformIO project!
Error: The package 'toolchain-gccarmlinuxgnueabi' is not available for your system 'windows_amd64'

One of the boards led to the mistaken attempt to use the LInux form of GCC.  
Could be ESP8266, could be Teensy ARM Cortex, don't know.

#### Steps to Reproduce

1.New install. Choose boards including ESP8266, Teensy 3.x boards, STM32F boards (3)
2.
3.

### Actual Results
PlatformIO: Failed to initialize PlatformIO project!
Error: The package 'toolchain-gccarmlinuxgnueabi' is not available for your system 'windows_amd64'
### Expected Results
Successful INITIALIZATION

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

childresss, 2016-06-02:
Choosing Raspberry Pi 2 or Raspberry Pi Zero causes the error reported in this usse.

ivankravets, 2016-06-15:
Duplicate of https://github.com/platformio/platformio/issues/578

P.S: We don't have currently toolchain for cross compilation to RaspberryPi. Please use PlatformIO directly on RPi. PlatformIO CLI works very well on RPi.

Dorry for the inconveniences.

closed 2016-06-15 by ivankravets