platformio/platformio-core #682: Support for ESP32

markbeee, 2016-06-02:
Will boards based on Espressif ESP32 supported soon or is possible to install it manually?

### Configuration

**Operating system**:

WIndows, Linux, OSX

### Additional info

More information on Espressif Website: http://espressif.com/products/hardware/esp32/resources

Testato, 2016-10-05:
+1

the ESP32 is officially out, so now is time to add it :-)

svenp, 2016-10-06:
The Core:
https://github.com/espressif/arduino-esp32
:-)

ekarlso, 2016-10-16:
Is there any news on this or ?

forthlightning, 2016-11-02:
need moar boards

bmv437, 2016-11-03:
Looks like they are working on it. https://github.com/platformio/platform-espressif32
@ivankravets any ETA?

ivankravets, 2016-11-03:
We are waiting for ESP32 Dev Kits from Espressif System to test it. Espressif System has alerady shipped. 1-2 weeks...

Testato, 2016-11-03:
Will platformio support the arduino core for esp32 and the espressif sdk also ?

ivankravets, 2016-11-03:
Sure, see https://github.com/platformio/platform-espressif32/blob/develop/platform.json#L23:L32

blackketter, 2016-11-05:
Is this worth trying out yet?  I have a SparkFun ESP32 Thing here and would be willing to test...

blackketter, 2016-11-05:
Well, to answer my own question, on PlatformIO, version 3.1.0, I tried installing platform-espressif32 manually and a blink sample builds but fails to link like this:

```
lucky13:esp32 dean$ pio run
[Sat Nov  5 09:39:30 2016] Processing esp32thing (framework: arduino, lib_deps: , platform: espressif32, upload_speed: 115200, board: esp32dev, upload_port: esp32thing.local)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 10 compatible libraries
Looking for dependencies...
Project does not have dependencies
Linking .pioenvs/esp32thing/firmware.elf
/Users/dean/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.literal+0x24): undefined reference to `esp_setup_syscalls'
/Users/dean/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.literal+0x28): undefined reference to `esp_vfs_dev_uart_register'
/Users/dean/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.literal+0x2c): undefined reference to `esp_reent_init'
/Users/dean/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libesp32.a(cpu_start.o): In function `start_cpu0_default':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:153: undefined reference to `esp_setup_syscalls'
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:163: undefined reference to `esp_vfs_dev_uart_register'
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:165: undefined reference to `esp_reent_init'
/Users/dean/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libfreertos.a(tasks.o): In function `prvInitialiseNewTask':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c:4312: undefined reference to `esp_reent_init'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32thing/firmware.elf] Error 1
============================================================================ [ERROR] Took 1.61 seconds ============================================================================
```

ivankravets, 2016-11-05:
@blackketter need to sync https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L72 with https://github.com/espressif/arduino-esp32/blob/master/platform.txt

ivankravets, 2016-11-06:
Update! You can try inttial implementation by @valeros. See https://github.com/platformio/platform-espressif32

The build results: https://travis-ci.org/platformio/platform-espressif32

blackketter, 2016-11-06:
Awesome, blink builds, uploads and runs on my Sparkfun ESP32 Thing.  I do find that I need to manually reset it after an upload, will look into that...

ivankravets, 2016-11-19:
Docs: http://docs.platformio.org/en/latest/platforms/espressif32.html
Examples: https://github.com/platformio/platform-espressif32/tree/develop/examples

closed 2016-11-19 by ivankravets

vshymanskyy, 2016-11-20:
@ivankravets it looks like not all -D flags are set correctly: https://travis-ci.org/blynkkk/blynk-library/jobs/177357568

ivankravets, 2016-11-21:
@vshymanskyy Thanks! Fixed in https://github.com/platformio/platform-espressif32/releases/tag/v0.1.1

P.S: v0.1.1 is released, just restart a Travis.CI Job.

vshymanskyy, 2016-11-21:
Thank you! Fixed.

rolandoo, 2016-11-28:
Hi, how to fix this issue? Thanks.

[image]

commit f7790ce5 (2016-11-19): Add support for Espressif 32 // Resolve #682