platformio/platformio-core #683: SPL for F0 variant

zserge, 2016-06-02:
I'm trying to use PlatformIO with custom STM32F0 boards (F051 and F030). Previously I've been using libopencm3 framework, all worked fine, but now I need SPL as a framework.

It turns out that SPL installed by PlatformIO comes with F4, F3 and L1 variants. Would it be possible add missing variants, too, so people could use all STM32 MCUs?

zserge, 2016-06-05:
I tried to re-pack the framework-spl and replace it manually. It seems to do the trick, but I don't know how to automate it or offer a pull request. Attached it my hand-crafted package with SPL for F0 included.

[framework-spl.tar.gz](https://github.com/platformio/platformio/files/299212/framework-spl.tar.gz)

valeros, 2016-06-09:
Updated, thanks!

closed 2016-06-09 by valeros

zserge, 2016-11-28:
Sorry @ivankravets but I would like to re-open this.

The assembly file `framework-spl/stm32/cmsis/variants/stm32f0/startup_stm32f0xx.s` is named with a lower-case `.s`, causing build problems. Renaming to the upper-case `.S` helps, but it should better be renamed in the SPL F0 package.

ivankravets, 2016-11-28:
@zserge Please open new issue here ( https://github.com/platformio/platform-ststm32/issues ) with further explanations what need to do. Thanks.