platformio/platformio-core #684: Add support for 8-bit PICs through SDCC and gputils

tweakyllama, 2016-06-03:
Would it be possible to add support for 8-bit pics by creating toolchains for [SDCC](http://sdcc.sourceforge.net/) and [gputils](https://sourceforge.net/projects/gputils/). Or is this possible through a custom platform?

shree-0403, 2018-01-02:
Is there any improvement on this question.I am thinking to use platformio for pic16. Or any direction How to use it.

ivankravets, 2018-03-13:
Duplicate of https://github.com/platformio/platformio-core/issues/704

closed 2018-03-13 by ivankravets

shree-0403, 2018-03-13:
Thank you for reply.
But what about answer, Can I use platformio for pic16. Or any direction How
to use it.

devrim-oguz, 2020-11-20:
This needs to happen already! +1 For the PlatformIO Pic support

mario2788, 2021-05-10:
I invite you to know JAL to work all 8-bit MCUs of michrochip:
http://justanotherlanguage.org/
https://www.youtube.com/watch?v=J3Vok7s2fXk&t=3s

There is also a core for JAL in Visual Studio:

Requirements & Installation Process:

1- Download,install and run Visual studio code from  https://code.visualstudio.com/download
2- Press Ctrl-Shit-X to invoke the Extensions and search for JAL (or trying to open a JAL file will also prompt to install the extension)
3- Click Install to add the extension to vscode
4- Set the path to your jal compiler exe and library file path by clicking the gear icon in extension manager and selecting 
5- Extension Settings (It's also possible by going to editor preferences and searching for JAL)
6- Press Ctrl-Shift-P and Type/select "Tasks:Open User Tasks''.
7- Select Others (Example to run arbitrary command).
      In the opened tasks.json paste the following code

    {
        // prefilled tasks.json for compiling a JAL file
        "version": "2.0.0",
        "tasks": [
            {
                "label": "Compile JAL File",
                "type": "process",
                "command": "${config:jal.paths.exePath}",
                "args": [
                    "${file}",
                    "-s",
                    "${config:jal.paths.LibPath}"
                ],
                "presentation": {
                    "reveal": "always",
                    "panel": "new"
                },
                "problemMatcher": [],
                "group": {
                    "kind": "build",
                    "isDefault": true
                }
            }
        ]
    }

8- Open any JAL file and press Ctrl-Shift-B to build the JAL file and you are ready to rock !

infinity-computers-dot-net, 2021-05-10:
> I invite you to know JAL to work all 8-bit MCUs of michrochip:
> http://justanotherlanguage.org/

[Great Cow Basic](http://gcbasic.sourceforge.net/Typesetter/index.php/Home) would be a superior alternative.

devrim-oguz, 2021-05-10:
But maybe these can be added to PlatformIO as different platforms

WebDust21, 2023-09-25:
> > I invite you to know JAL to work all 8-bit MCUs of michrochip:
> > http://justanotherlanguage.org/
> 
> [Great Cow Basic](http://gcbasic.sourceforge.net/Typesetter/index.php/Home) would be a superior alternative.

I tried it (GCB).  Started out pretty decent--but I ran into inconsistencies with the infrastructure (in particular, how the UARTs were handled).  And the head guy effectively slammed the door in my face when I pointed them out.  Treating yesterday's mistakes as sacrosanct pretty well cured me of using GreatCowBasic.

Anobium, 2023-09-25:
Re GC BASIC.

I am the head guy. 

Can you share what my response was? I honestly do not remember slamming the door on USART. If I did then I apologise and I will take the feedback.

Ping me a PM.

GC BASIC supports PICs, AVRs and LGT and the designed intent is to support USART with hardware solutions and optimised software solutions.

WebDust21, 2023-09-25:
@Anobium https://sourceforge.net/p/gcbasic/discussion/1073404/thread/2f06767845/?limit=25#5987
That's one reference.  IIRC there was another one (which I'm having trouble finding, as that was 4 years ago) that ended up with some reference to the original creator of GCB not wanting to change something, with the resultant excuse that "that's just how he is."

Found that Microchip XC8 was free shortly thereafter, and have been using that ever since.

Anobium, 2023-09-25:
> @Anobium https://sourceforge.net/p/gcbasic/discussion/1073404/thread/2f06767845/?limit=25#5987 That's one reference. IIRC there was another one (which I'm having trouble finding, as that was 4 years [...]

That was a 4 years ago.  Sorry.

We have actually taken on-board the approach since then and the delays can be remove with ease.  And, the library is now public and it can be updated via SVN.

Also, Hugh, the GC BASIC creator, has not been part of the project for many years, and, we have now moved on.  We are now have migrated to VSCode, we have full support for XC8 and we can integrate and interoperate with MPLAB-X.

WebDust21, 2023-09-25:
Also worth noting that I'm moving away from Microchip PICs entirely, as they're quite expensive, went completely out of stock/unattainable during the chip shortage (often with 80-week lead times)...and not that powerful for the price.  
Now I'm using the Nuvoton MS51-series chips--and for goodness sakes, they knock the pants right of any of the Microchip PICs I ever used (PIC10, PIC12, PIC16, PIC18, dsPIC33).  Much cheaper, much more powerful, and CONSIDERABLY more power efficient.  Not to mention a lead time of ~6 weeks--and they never went out of stock during the chip shortage.  AND I don't have to use Buglab IDE, either...

> and the delays can be remove with ease.

And therein's the root of the problem: the "delays" were a Band-aid fix to a very poorly written UART handler (which was simply not using the MCU-supplied "transmit complete" flag).  Having a fixed "byte send delay" on a UART handler which can easily be set to a different baud rate (i.e. different actual byte send duration) is simply quite laughable.
AND "solving the problem" by giving the user the option to apply a Band-aid to a fully-known problem--instead of fixing the root problem--is a _very_ lousy solution in my book.

Anobium, 2023-09-25:
> Also worth noting that I'm moving away from Microchip PICs entirely, as they're quite expensive, went completely out of stock/unattainable during the chip shortage (often with 80-week lead times)...and not that powerful for the price. Now I'm using the Nuvoton MS51-series chips--and for goodness sakes, they knock the pants right of any of the Microchip PICs I ever used (PIC10, PIC12, PIC16, PIC18, dsPIC33). Much cheaper, much more powerful, and CONSIDERABLY more power efficient. Not to mention a lead time of ~6 weeks--and they never went out of stock during the chip shortage. AND I don't have to use Buglab X, either...
> 
> > and the delays can be remove with ease.
> 
> And therein's the root of the problem: the "delays" were a Band-aid fix to a very poorly written UART handler (which was simply not using the MCU-supplied "transmit complete" flag). Having a fixed "byte send delay" (IIRC) on a UART handler which can easily be set to a different baud rate (i.e. different actual byte send duration) is simply quite laughable. AND "solving the problem" by giving the user the option to apply a Band-aid to a fully-known problem--instead of fixing the root problem--is a _very_ lousy solution in my book.

Lots of agreement here.

Hence, I added LGT to GCBASIC. Twice as fast as AVR, less power, way more capable chipset.  

We, GCBASIC team, rewrote the USART library in 2021. Biil Roth took the whole library and reworked.

Nuvoton is on the list of support chip families. Z80 is underway.  Just need time to progress development.

WebDust21, 2023-09-25:
> Nuvoton is on the list of support chip families.

Nuvoton's provided BSPs are bent towards proprietary and expensive tooling (IAR and Keil)--but not too terribly difficult to rework for SDCC.  My reworking has been ad-hoc at best, but it DOES work.
Worth noting that PlatformIO already has an "mcs51" toolchain with entries for pretty much all of the Nuvoton MS51-series chips--and it works pretty good, although doesn't appear to come with the Nuvoton BSP AFAIK.  (Only found the MCS51 entry for PlatformIO about three weeks ago.)

The major weak link with the Nuvoton offerings, is that they don't have programmer software for Linux/MacOS.
Someone wrote a tool called "nuvoprog" for the Nu-Link-Pro programmer, supporting the N76E Nuvoton MCU series...and another person forked it and added MS51FB9AE support (https://github.com/jdelphi/nuvoprog2).  I've since added MS51TC0AE to my copy (not on GH)--but that tool desperately needs rewritten in something better than Go, and with a USB library that isn't marked EOL/deprecated for MacOS.

Anobium, 2023-09-25:
> Nuvoton's provided BSPs are bent towards proprietary and expensive tooling (IAR and Keil)--but not too terribly difficult to rework for SDCC. My reworking has been ad-hoc at best, but it DOES work. [...]

We can look into this.

We develop and sell PICKITPLUS software. So, we have the toolchain to make changes to support these chips.  It would depend on market and RoI.

infinity-computers-dot-net, 2023-09-26:
@WebDust21 These Nuvoton chips sound interesting. What's your experience with PlatformIO been like with those, thus far? Does it work well?

Also, what programming hardware is needed? And is the software for that available free of charge?

What is a "BSP"?

Cheers.

WebDust21, 2023-09-26:
> What's your experience with PlatformIO been like with those, thus far? Does it work well?

Well, I started out programming them effectively with Notepad (i.e. Kwrite or Gedit in Linux), and a Bash script for compilation (based on a Windoze BATch script from the Nuvoton BSP IIRC).  This of course means that the PlatformIO / VSCode infrastructure has been a HUGE improvement ;-).  The biggest quirk/complaint I have is that [VSCode does not support SDCC keywords](https://github.com/Microsoft/vscode-cpptools/issues/2499)--and as a result, VSCode's syntax highlighter thinks that my perfectly-compiling Nuvoton MS51 project has over 168 errors in it.  
Apart from that, however, being able to jump back and forth in the project by keywords/functions/defines, having auto-complete, etc., etc. is a real bonus compared to Notepad!

> Also, what programming hardware is needed?

I personally am using a Nu-Link-Pro, from Nuvoton.  You can buy it from [direct.nuvoton.com](https://direct.nuvoton.com/en/nu-link-pro), or [you can also purchase it from Digi-Key](https://www.digikey.com/en/products/detail/nuvoton-technology-corporation/nu-link-pro/3065247).  It supports almost all of the Nuvoton MCU lineup, including the ARM-core processors that I am hoping to get into next (if I can figure out how to hack an M23 system into a PlatformIO setup...)

> And is the software for that available free of charge?

Yes, absolutely.  If you're on Windoze, you're quite in luck: download the Nuvoton ICP Tool (_*not*_ the ISP tool!) from Nuvoton's website.  Direct link: https://www.nuvoton.com/resource-download.jsp?tp_GUID=SW1720200221181328_
(software link: https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/index.html)
For Linux / OSX, you're currently limited to the aforementioned Nuvoprog--which is an absolute pain to compile due to being written in Go.
I'll be happy to provide the compiled Nuvoprog binary for Linux if you so need.  It currently supports the N76E003 (which I've never used), MS51FB9AE and MS51TC0AE.

> What is a "BSP"?

Board Support Package.
It is effectively a Microchip ".inc" file (processor special-function register definitions) + SFR manipulation macros + some basic subroutines for setting up the peripherals.

Worth noting that the Nuvoton processors aren't for the faint of heart...there's very little community around them, and finding the proper Technical Reference Manual can be a bit of a challenge.  But if you're quite familiar with low-level programming (especially in assembler!), the sky is the limit with those amazing processors.  
As far as "power efficient", I'm thinking of the Microchip dsPIC30F4013 that I was running at 30MIPs (= 120MHz).  The 40-pin DIP actually got quite warm to the touch...at 30MIPs.  Was pulling 120mA @ 5v, too (=0.6W).  Whereas these Nuvoton MS51 chips run an 8051 1T core (as low as one clock cycle per instruction)--so they can effectively run past ~20MIPs (@ 24MHz), and pull less than 5mA @ 5v while doing so.  (Yes, some Microchip PIC18s run a 3.3v internal core voltage, and run up to 16MIPs [=64MHz] at ~15mA.  I'm thinking of the PIC18F6xK22/PIC18F8xK22.)

I mean, the MS51 processors allow you to programmatically reconfigure the second UART (wait, that means the chip comes with TWO!  Take _that_ Microchip!)...to none other than the ICP pins!  This makes it incredibly easy to use a single "ICSP" header to program the chip...and then swap out the Nu-Link-Pro for a USB-serial converter and use the ICSP port as a debug port.  (I have never done ICD-style work...as far as I got was totally and irretrievably bricking a Microchip PIC16F747 by attempting to do debug stuff with MPLAB IDE + an ICD2 clone.  Half my fault admittedly!)
Oh, and you can also programmatically reconfigure an I2C controller to the ICP pins on an MS51 Nuvo!  I admittedly did just that on a project: I hardwired the ICP pins of _two_ Nuvoton MS51 chips together, as well as to the programming pins of an ESP32.  Not only can I program the ESP32 from the ICSP header, I could also program **both** Nuvoton chips simultaneously (!)--AND then once the user programs were running, the ESP32 could communicate with either Nuvoton MS51 via I2C (over the exact same pins).  (I used a pin on each Nuvo as an "ID" pin--one was tied high, the other tied low.  That way they could self-set different addresses.)

Nuvoton MS51 ADCs are also quite incredible: not only are they 12-bit (*most* PICs are only 10-bit), they are also rated to scream right up to 500Ksps without hesitation.  (You effectively have to overclock a Microchip PIC ADC to reach the rated 100Ksps speed.)  AND the MS51 chips have a band-gap channel on the ADC.  (Most PICs don't.)  PLUS factory-calibrated band-gap correction stored in the memory, so you can make accurate readings with it right out the gate.  (PICs don't.  And boy is the band-gap error on PICs terrible!  I know...because I tried using it on a project.)
AND the MS51s come with a unique ID burned into memory.  (Not on the PIC10/12/16/18/dsPIC30.)
Plus self-write capability (many PICs don't), as well as bootloader provisions (most PICs don't).

And I haven't gotten into a quarter of the features ;-).  I'd better quit here while I'm ahead!

infinity-computers-dot-net, 2023-09-30:
@WebDust21
Wow! That's a lot of information. Thank you for taking the time to explain it to me.

My main concerns from what you've told me are (a) VSCode showing errors where there are none, (b) the implication that I would need to program them in assembler, and (c) the cost of a Nu-Link-Pro.

Would [this item](https://vi.aliexpress.com/item/33027792033.html) be suitable?

I have never written anything in assembler, although I understand the concept. The lowest I have gone is C. I very much doubt I would be very productive in asm.

WebDust21, 2023-09-30:
> (a) VSCode showing errors where there are none

VSCode's fault--and M$FT doesn't seem too interested in adding SDCC nomenclature support.  Oh well.  PlatformIO supports other IDEs, one of which _might_ have better support for SDCC nomenclature...

> (b) the implication that I would need to program them in assembler

Oh no, absolutely not.  SDCC = Small Device C Compiler.
My comment about needing to be familiar with low-level programming was for debug purposes (as well as simply being versed with the processor's specific features/abilities/quirks).  For example, every now and then I run into issues where the processor doesn't do what I expected it to--and after exhausting every possible option in the source code, my last resort is to dig into the LST (assembly list) output file, search for the offending "C source" line, and check the assembler output.  (But you have to be somewhat versed in assembler for this to be of any value!)
Most recently was actually with SDCC, where a subtract between two 8-bit "unsigned char" registers wasn't "rolling over" like I expected.  Dug into the LST file, only to find that SDCC had decided to change the mathematics (between two unsigned char registers, mind you!) to a 16-bit signed int subtract.  Of _course_ it didn't work properly!
Upon finding that, I moved the subtract to a separate line, i.e. unsigned char foo = op1 - op2; --and then do the necessary check in the IF.  Problem solved.

It's not just SDCC that does this--I've had to do it countless times with XC8 in MPLAB X.  Generally speaking, it seems the more mathematics I try to cram into a statement, the greater mess the compiler makes.

> (c) the cost of a Nu-Link-Pro.

You aren't going to find a cheaper OEM programmer than the Nu-Link-Pro at ~$40.  Heck, an MPLAB PicKit4 is >$200 if purchased OEM (i.e. not a Chinese knockoff with varying functionality).
The linked programmer...I have no idea, as I've never used one.  It might work...maybe.  (I've only used the OEM Nu-Link-Pro.)  I wouldn't try it if you're on Linux or MacOS, as the chances of it not working with "nuvoprog" are pretty high.