platformio/platformio-core #685: st-flash fails on 1st upload (stlink)

zgoda, 2016-06-03:
Built in st-flash fails to upload firmware when run for 1st time, 2nd consecutive attempt succeeds.

```
$ pio run -t upload
[Fri Jun  3 17:03:05 2016] Processing disco_f051r8 (upload_protocol: stlink, board: disco_f051r8, framework: mbed, platform: ststm32)
------------------------------------------------------------------------------------------------------------------------
"/home/jazg/.platformio/packages/tool-stlink/st-flash" write .pioenvs/disco_f051r8/firmware.bin 0x08000000
2016-06-03T17:03:05 INFO src/stlink-usb.c: -- exit_dfu_mode
2016-06-03T17:03:05 INFO src/stlink-common.c: Loading device parameters....
2016-06-03T17:03:05 INFO src/stlink-common.c: Device connected is: F0 device, id 0x20006440
2016-06-03T17:03:05 INFO src/stlink-common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2016-06-03T17:03:05 INFO src/stlink-common.c: Attempting to write 20208 (0x4ef0) bytes to stm32 address: 134217728 (0x8000000)
2016-06-03T17:03:06 INFO src/stlink-common.c: Finished erasing 20 pages of 1024 (0x400) bytes
2016-06-03T17:03:06 INFO src/stlink-common.c: Starting Flash write for VL/F0/F3 core id
2016-06-03T17:03:06 INFO src/stlink-common.c: Successfully loaded flash loader in sram
2016-06-03T17:03:08 ERROR src/stlink-common.c: flash loader run error
2016-06-03T17:03:08 ERROR src/stlink-common.c: run_flash_loader(0x8000000) failed! == -1
Flash page at addr: 0x08004c00 erased
stlink_fwrite_flash() == -1
scons: *** [upload] Error 255
============================================= [ ERROR ] Took 3.49 seconds =============================================
jazg@JAZG-LENOVO:~/work/stm32_lcd $ pio run -t upload
[Fri Jun  3 17:03:18 2016] Processing disco_f051r8 (upload_protocol: stlink, board: disco_f051r8, framework: mbed, platform: ststm32)
------------------------------------------------------------------------------------------------------------------------
"/home/jazg/.platformio/packages/tool-stlink/st-flash" write .pioenvs/disco_f051r8/firmware.bin 0x08000000
2016-06-03T17:03:18 INFO src/stlink-common.c: Loading device parameters....
2016-06-03T17:03:18 INFO src/stlink-common.c: Device connected is: F0 device, id 0x20006440
2016-06-03T17:03:18 INFO src/stlink-common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2016-06-03T17:03:18 INFO src/stlink-common.c: Attempting to write 20208 (0x4ef0) bytes to stm32 address: 134217728 (0x8000000)
2016-06-03T17:03:19 INFO src/stlink-common.c: Finished erasing 20 pages of 1024 (0x400) bytes
2016-06-03T17:03:19 INFO src/stlink-common.c: Starting Flash write for VL/F0/F3 core id
2016-06-03T17:03:19 INFO src/stlink-common.c: Successfully loaded flash loader in sram
Flash page at addr: 0x08004c00 erased
2016-06-03T17:03:20 INFO src/stlink-common.c: Starting verification of write complete
2016-06-03T17:03:20 INFO src/stlink-common.c: Flash written and verified! jolly good!
 19/19 pages written
============================================= [SUCCESS] Took 2.41 seconds =============================================
```

Standalone st-flash utility does not have this problem.

ivankravets, 2016-06-15:
@valeros Should we update our `st-link` package?

valeros, 2016-06-17:
Hi @zgoda !
`tool-stlink` package was updated to `1.2.0` version. Could you please re-test it in your system?
Thanks!

zgoda, 2016-06-21:
Now it works in CLI, no change in IDE. Package tool-stlink is at version 3 in both environments.

zgoda, 2016-06-21:
Actually it works in IDE as well, had to restart after upgrading tool-stlink. Thank you guys.

closed 2016-06-21 by ivankravets