platformio/platformio-core #691: [Bug report] Cannot upload firmware to Arduino Leonardo. Port enumerated too quickly

maziu, 2016-06-11:
### Configuration

**Operating system**: Windows 7 64 bit

**PlatformIO Version** (`platformio --version`): 2.9.4

### Description of problem

Cannot upload software to Arduino Leonardo. Instead of waiting for new port, avrdude is called with old port parameter.

#### Steps to Reproduce

1. Connect arduino leonardo
2. Try to upload any example project

### Actual Results

`
Auto-detected UPLOAD_PORT/DISK: COM7
Waiting for the new upload port...
"C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude" -v -p atmega32u4 -C "C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude.conf" -c avr109 -b 57600 -P "COM7" -D -U flash:w:.pi
oenvs\leonardoeth\firmware.hex:i
 
avrdude: Version 6.0.1, compiled on Apr  3 2014 at 23:12:16
System wide configuration file is "C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude.conf"
MS-DOS style path detected: C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude.conf
Preferred POSIX equivalent is: /tool-avrdude/avrdude.conf
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch
 
cygwin warning:
CYGWIN environment variable option "nodosfilewarning" turns off this warning.
Consult the user's guide for more details about POSIX paths:
http://cygwin.com/cygwin-ug-net/using.html#using-pathnames
 
avrdude: ser_open(): can't set com-state for "\\.\COM7"
avrdude: ser_send(): write error: sorry no info avail
Using Port                    : COM7
Using Programmer              : avr109
Overriding Baud Rate          : 57600
scons: *** [upload] Error 1
`

In device manager i see that COM7 disappears and for moment COM9 appears. SW should be uploaded to COM9.
From arduino IDE it works fine, avrdude is called with COM9 parameter.

###

I fixed it by modifying platformio\builder\tools\pioupload.py , function WaitForNewSerialPort. I added 1 second delay at beginning of this function, so there's time for port to enumerate, but I'm almost sure that this is not correct solution (i mean, it works, but it's just quickfix).
After this change it works fine (please look at function below)

`def WaitForNewSerialPort(env, before):
    print "Waiting for the new upload port..."
    prev_port = env.subst("$UPLOAD_PORT")
    new_port = None
    elapsed = 0
    sleep(1)
    while elapsed < 5 and new_port is None:
        now = get_serialports()
        for p in now:
            if p not in before:
                new_port = p['port']
                break
        before = now
        sleep(0.25)
        elapsed += 0.25

    if not new_port:
        for p in now:
            if prev_port == p['port']:
                new_port = p['port']
                break

    if not new_port:
        env.Exit("Error: Couldn't find a board on the selected port. "
                 "Check that you have the correct port selected. "
                 "If it is correct, try pressing the board's reset "
                 "button after initiating the upload.")

    return new_port
`

ivankravets, 2016-06-11:
Thanks a lot for the report.

1. Please disable your hook with `# sleep(1)`
2. Try to increase timeout from 5 sec to 10sec. `while elapsed < 10 and new_port is None:`

Does it work for you?

maziu, 2016-06-11:
I did it.
No, it does not help. 
This lines
```
Waiting for the new upload port...
"C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude" -v -p atmega32u4 -C "C:\Users\Mateusz\.platformio\packages\tool-avrdude\avrdude.conf" -c avr109 -b 57600 -P "COM7" -D -U flash:w:.pioenvs\leonardoeth\firmware.hex:i
```
are executed directly one after other (maybe 50-100ms delay)

ivankravets, 2016-06-11:
Please re-test the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-06-11 by ivankravets

valabrijesh, 2017-04-02:
avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't set com-state for "\\.\COM9"

avrdude done.  Thank you.
 
Code is not uploaded on arduino uno board. Help me

robertguyser, 2019-12-09:
I am having this problem in December 2019. Is there any advice on how to fix it?

ivankravets, 2019-12-09:
Do you mean this issue https://github.com/platformio/platform-atmelavr/issues/172?

maziu, 2019-12-10:
@robertguyser 
Did you try my quickfix with  sleep(1) before while?

commit 3b8bc422 (2016-06-11): Improve firmware uploading to Arduino Leonardo based boards // Resolve #691
commit 36a9bf69 (2016-06-14): Merge branch 'develop' into feature/platformio-30

* develop:
  Add support for ST Nucleo L031K6 board to ARM mbed framework
  Fix typo
  Fix pylint error
  Update history
  Improve linker flags handling for mbed
  Update examples
  Improve include selection for mbed boards with a custom vendor
  Re-submit a package to PyPI
  Version bump to 2.10.0 (issues #683, #687, #688, #689, #691)
  Pass PlatformIO macros with version to libs
  Update ARM mbed framework package to v121
  Typo fixes
  Move custom platform and board to separate page
  Add new ST Nucleo board
  Update build script for mbed f [... cut]