platformio/platformio-core #692: platformio deplibs with extra_script

wuwx, 2016-06-12:
I have manage a project with platform, the project struct like this:
```
src:
  user_main.c
lib:
  hello
    hello.c
    hello.h
```

user_main.c:
```
    #include "hello.h"
    void user_init(void)
    {
        printf("Hello\r\n");
    }
```
platformio.ini:
```
    extra_script = extra_script.py
```

extra_script.py:
```
    env.Append(
    
        CPPPATH=[
           join("$PROJECT_DIR", "sdk", "include"),
        ]
    )
```

but when i use ```pio run``` build the project, lib/hello doesn't include sdk/include path

but user_main.o include sdk/include

it seems lib directory can't support extra_script

how can i fix it, thanks

ivankravets, 2016-06-12:
This is not possible with PlatformIO 2.0. PlatformIO builds libraries in isolated environments. You will be able to use `extraBuildFlags` in PlatformIO 3.0.

Temporary solution is to use `build_flags = -I$PROJECT_DIR/sdk/includes`

ivankravets, 2016-06-12:
Related issues

* https://github.com/platformio/platformio/issues/542
* https://github.com/platformio/platformio/issues/289

closed 2016-06-12 by ivankravets