platformio/platformio-core #693: upload fail: rpi b wheezy to arduino nano

doktorsuarez, 2016-06-12:
hi there
installing platformio and building works fine!
when i try to upload the sketch i get this error:

```
BeforeUpload(["upload"], [".pioenvs/nanoatmega328/firmware.hex"])
scons: *** [upload] GetSerialPortsError : No implementation for your platform ('posix') available
Traceback (most recent call last):
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Action.py", line 1063, in execute
result = self.execfunction(target=target, source=rsources, env=env)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/scripts/atmelavr.py", line 68, in BeforeUpload
before_ports = get_serialports()
File "/usr/local/lib/python2.7/dist-packages/platformio/util.py", line 287, in get_serialports
raise exception.GetSerialPortsError(os.name)
GetSerialPortsError: No implementation for your platform ('posix') available
```

My configuration is:

**Operating system**: Raspian wheezy

**PlatformIO Version** (`platformio --version`): PlatformIO 2.9.4 

traget system: Arduino Nano v3 - ttyUSB0

i specified this in the .ini file.

another strange thing is:
platformio serialports list return an error:

```
 sudo platformio serialports list
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Error: No such command "serialports"
```

ivankravets, 2016-06-12:
Does it work when use specify `upload_port` directly in `platformio.ini`?

ivankravets, 2016-06-12:
Also, please provide me an output of `pip list` command. Thanks,

doktorsuarez, 2016-06-12:
Thanks for the fast answer!
No, specifiying upload_port wasn't succesful.

pip list:
bottle (0.12.9)
click (5.1)
colorama (0.3.7)
configobj (5.0.6)
glob2 (0.4.1)
ino (0.3.7)
Jinja2 (2.8)
lockfile (0.12.2)
lxml (2.3.2)
MarkupSafe (0.23)
mindwavemobile (0.200)
numpy (1.6.2)
ordereddict (1.1)
pexpect (2.4)
picamera (1.10)
pifacecommon (4.1.2)
pifacedigitalio (3.0.4)
pigpio (1.17)
PIL (1.1.7)
pip (8.1.2)
platformio (2.9.4)
PyBluez (0.18)
pygame (1.9.1release)
pygobject (3.8.2)
pyserial (3.1)
reportlab (2.5)
requests (2.10.0)
RPi.GPIO (0.6.2)
setuptools (14.0)
six (1.10.0)
smbus (1.1)
wheel (0.29.0)
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

not sure what the warning wants points to

thx

ivankravets, 2016-06-14:
See comment here https://community.platformio.org/t/chromebook-python-serial-error/385/2?u=ivankravets

Please try it.

doktorsuarez, 2016-06-14:
Hi there,

one issue is:
python-serial package for wheezy doesn't contain serial.tools.list_ports module.
https://bugs.launchpad.net/raspbian/+bug/1269860

so i updated to raspian jessie, now everything works fine!

thx

closed 2016-06-14 by doktorsuarez