platformio/platformio-core #694: platformio 2.10.0 cannot display help

veritazz, 2016-06-13:
Hi, I just upgraded but something seems odd. When I enter:

```
platformio --help
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Error: No such command "test"
```

Also *platformio upgrade* did fail with the following issue:

```
# platformio upgrade
Please wait while upgrading PlatformIO ...
Error: ['/usr/bin/python', '-m', 'pip', '--no-cache-dir', 'install', '--upgrade', 'platformio']

Usage:   
  /usr/bin/python -m pip <command> [options]

no such option: --no-cache-dir

* Upgrade using `pip install -U platformio`
* Try different installation/upgrading steps:
  http://docs.platformio.org/en/latest/installation.html
```

I am on Ubuntu 14.04 LTS.

ivankravets, 2016-06-13:
Please sorry, I've just re-submitted package to PyPi as 2.10.1.

Please upgrade to it.

veritazz, 2016-06-13:
The *--version* works now but *--help* still seems not to work.

# platformio --help
Usage: platformio [OPTIONS] COMMAND [ARGS]...

Error: No such command "test"
# platformio --version
PlatformIO, version 2.10.1
veritazz@Schleppi:~/projects/Arduino/Fatsche$

ivankravets, 2016-06-13:
Try:
```
pip uninstall platformio
pip install -U platformio
```

ivankravets, 2016-06-13:
Please reopen if you still need help

closed 2016-06-13 by ivankravets

veritazz, 2016-06-13:
Many thanks. That solved the issue.

ivankravets, 2016-06-13:
Please sorry for the inconvenience! 

Happy Coding with PlatformIO!