platformio/platformio-core #695: Building not possible, because constants like PI or OUTPUT can't be found

tennismichel, 2016-06-13:
------------------------------------------------------------------

### Configuration

**Operating system**: OSX 10.11.4

**PlatformIO Version** (`platformio --version`): latest

### Description of problem
When I want to build my project or any other project that worked previously, I get a lot of errors that all look kind of like the following..

.pioenvs/genuino101/FrameworkArduino/WCharacter.h:43:8: error: 'boolean' does not name a type
inline boolean isHexadecimalDigit(int c) attribute((always_inline));

.pioenvs/genuino101/FrameworkArduino/WCharacter.h:135:8: error: 'boolean' does not name a type
inline boolean isUpperCase(int c)

Additionally I can see that constants like PI or OUTPUT for the PIN Mode are not found or recognized.

The code shouldn't be the problem. It's working on other Mac's with Platformio and it has been working with mine earlier, too. Tried to reinstall the whole IDE, but it's still not working. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:** 
[env:genuino101]
platform = intel_arc32
framework = arduino
board = genuino101

**Source file to reproduce issue:**
-main:
[Quadro_main.txt](https://github.com/platformio/platformio/files/312768/Quadro_main.txt)
-one of  the header files:
[RotorControl.txt](https://github.com/platformio/platformio/files/312769/RotorControl.txt)

-main again:

#include "Arduino.h"
#include <Timer.h>
#include "sensorfuse.h"
#include "PIDControl.h"
#include "RotorControl.h"
#include "HeightControl.h"
#include "ConstantData.h"

//declare Timer object
Timer t;
IMU imu;
PIDControl pidController;
RotorControl rotors;

//declare necessary Variabels
float targetPosition[3] = {0,0,0}; //reference_angle
float targetHeight;

void timerUpdate(){
  imu.update(SAMPLE_RATE);
  pidController.update(rotors.RotorSignal, imu.rot, imu.rot_vel, targetPosition, SAMPLE_RATE);
  rotors.update();
}
void updateSerial(){
}

void setup(){
  //Start Serial and wait for connection
  Serial.begin(38400);
  while(!Serial);

   //calibrate sensors
  imu.startAndCalibrate();
  pidController.set(); //calibrate PID_Regler to forget integrated sum (I)
  pidController.initializeSerialConnection();
  rotors.initialize(); //set Rotors and ESCs to PINs and initialize
  rotors.start();

  //Set timer event, that calls updateIMU every SAMPLE_RATE milliseconds
  t.every(SAMPLE_RATE,timerUpdate);
}

void loop(){
  t.update();
}

void serialEvent(){
  pidController.setConstantsViaSerial();
}

### Additional info

tennismichel, 2016-06-13:
last four lines of error message:

scons: *** [.pioenvs/genuino101/FrameworkArduinoVariant/variant.o] Error 1
scons: *** [.pioenvs/genuino101/FrameworkArduino/CDCSerialClass.o] Error 1
scons: *** [.pioenvs/genuino101/FrameworkArduino/IPAddress.o] Error 1
scons: *** [.pioenvs/genuino101/src/Quadro_main.o] Error 1

ivankravets, 2016-06-14:
> It's working on other Mac's with Platformio and it has been working with mine earlier, too

1. Do you use the same PlatformIO CLI version? To check use `pio --version`
2. Do you use the same sour code?

tennismichel, 2016-06-14:
Yes and yes. Cli version is 2.10.1. I thought the error messages may look familiar to you so I uploaded them, but we are sure it's not the code. It has to be something with the platformio Ide or Atom. 
To look at it from another perspective... This happened: I established a serial connection between my genuino 101 and matlab to plot some sensor data. It worked correct but after closing the connection and then opening another project, that worked earlier, to upload it to the arduino I got the errors from above. Since that all other projects aren't building as well. Thanks again for your help. Really appreciate and need it ;)

tennismichel, 2016-06-14:
Still have the same problem even after deleting all these files and then reinstalling: 

Delete the app from the Applications folder
rm -rf ~/.atom
rm /usr/local/bin/atom
rm /usr/local/bin/apm
rm ~/Library/Preferences/com.github.atom.plist
rm ~/Library/Application Support/com.github.atom.ShipIt
rm -rf ~/Library/Application Support/Atom/

ivankravets, 2016-06-14:
Ok, let's resolve this issue "step-by-step". I propose to start from PlatformIO CLI.

1. `rm -rf `~/.platformio`
2. `sudo pip install virtualenv`
3. `cd /tmp`, then `virtualenv piotest`
4. `source piotest/bin/activate`
5. `pip install platformio`
6. `pio --version` - should be 2.10.1
7. `pio run -d /path/to/your/project`

I'm waiting for the results. Thanks.

tennismichel, 2016-06-14:
Hey. It's almost working I think. After the first step it told me there isn't such a directory and that the reason for that might be that I would have to update Atom to version 8 that got out a few days ago. I just continued. 
The other steps worked well and after installing the Timer lib again, i was able to build the project without any problems. The only thing now is that I can't upload the Project to my Arduino. It's saying that Arduino is not responding and that I should try a Master reset, but unfortunately that's not helping.
Thank you so much for helping me out!!

ivankravets, 2016-06-14:
Could you provide an output from the uploading process using PlatformIO IDE?

tennismichel, 2016-06-14:
the last lines look like these:

arc-elf32-objcopy -S -O binary -R .note -R .comment -R COMMON -R .eh_frame .pioenvs/genuino
101/firmware.elf .pioenvs/genuino101/firmware.bin
MethodWrapper(["upload"], [".pioenvs/genuino101/firmware.bin"])
Looking for upload port/disk...
Auto-detected: /dev/cu.usbmodemFD121
"/Users/michaelbinder/.platformio/packages/tool-arduino101load/arduino101load" "/Users/mich
aelbinder/.platformio/packages/tool-arduino101load/dfu-util" .pioenvs/genuino101/firmware.b
in "/dev/cu.usbmodemFD121" verbose
Starting download script...
Args to shell: [/Users/michaelbinder/.platformio/packages/tool-arduino101load/dfu-util .pio
envs/genuino101/firmware.bin /dev/cu.usbmodemFD121 verbose]
Serial Port: /dev/cu.usbmodemFD121
BIN FILE .pioenvs/genuino101/firmware.bin
Waiting for device...
Waiting for device...
Waiting for device...
Waiting for device...
Flashing is taking longer than expected
Try pressing MASTER_RESET button
Waiting for device...
Waiting for device...
Waiting for device...
ERROR: Device is not responding.
scons: *** [upload] Error 1

ivankravets, 2016-06-14:
Thanks. The last, could you provide the same output from Arduino IDE?

tennismichel, 2016-06-14:
Well, I don't get an error message there. Uploading from Arduino IDE seems to work fine.

ivankravets, 2016-06-14:
Please go to Arduino IDE preferences (`CMD+,`) and check "Show verbose output during": "compilation" and "upload". Then please share here "upload" output. Thanks. Need to compare both results and find difference.

tennismichel, 2016-06-14:
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/michaelbinder/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/michaelbinder/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/michaelbinder/Documents/Arduino/libraries" -fqbn=Intel:arc32:arduino_101 -vid-pid=0X8087_0X0AB6 -ide-version=10609 -build-path "/var/folders/kl/d8p9z9213z5_pm6t97jgst8w0000gn/T/build04eca2f5a8e5cc36e20ff5baee3cdb00.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/michaelbinder/Documents/Arduino/DC_Motor_Potentiometer/src/DC_Motor_Potentiometer/DC_Motor_Potentiometer.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/michaelbinder/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/michaelbinder/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/michaelbinder/Documents/Arduino/libraries" -fqbn=Intel:arc32:arduino_101 -vid-pid=0X8087_0X0AB6 -ide-version=10609 -build-path "/var/folders/kl/d8p9z9213z5_pm6t97jgst8w0000gn/T/build04eca2f5a8e5cc36e20ff5baee3cdb00.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/michaelbinder/Documents/Arduino/DC_Motor_Potentiometer/src/DC_Motor_Potentiometer/DC_Motor_Potentiometer.ino"
"/Users/michaelbinder/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10609 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/common" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/drivers" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/bootcode" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/framework/include"  "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/cores/arduino" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/variants/arduino_101" "/var/folders/kl/d8p9z9213z5_pm6t97jgst8w0000gn/T/build04eca2f5a8e5cc36e20ff5baee3cdb00.tmp/sketch/DC_Motor_Potentiometer.ino.cpp" -o "/dev/null"
"/Users/michaelbinder/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10609 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/common" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/drivers" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/bootcode" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/framework/include"  "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/cores/arduino" "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/variants/arduino_101" "/var/folders/kl/d8p9z9213z5_pm6t97jgst8w0000gn/T/build04eca2f5a8e5cc36e20ff5baee3cdb00.tmp/sketch/DC_Motor_Potentiometer.ino.cpp" -o "/dev/null"
"/Users/michaelbinder/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10609 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-I/Users/michaelbinder/Library/Arduino15/packages/Intel/hardware/arc32/1.0.6/system/libarc32_arduino101/comm [... cut ...]
Download	[===============          ]  61%        12288 bytes
Download	[=================        ]  71%        14336 bytes
Download	[====================     ]  81%        16384 bytes
Download	[======================   ]  91%        17948 bytes
Download	[=========================] 100%        17948 bytes
Download done.
Sent a total of 17948 bytes
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
SUCCESS: Sketch will execute in about 5 seconds.

tennismichel, 2016-06-14:
it's a bit long, but I think the last part is the interesting one for you.. It's a different project, because I couldn't get the original to run in the Arduino IDE fast enough, because of the timer library. But this one gives the same output in Platformio as the original one.

ivankravets, 2016-06-14:
Hmm... Please open PlatformIO IDE and PlatformIO IDE Terminal and run this 
```
pio serialports list
```

Thanks.

tennismichel, 2016-06-14:
/dev/cu.Bluetooth-Incoming-Port
-------------------------------
Hardware ID: n/a
Description: n/a

/dev/cu.UEMEGABOOM-LWACP

------------------------
Hardware ID: n/a
Description: n/a

/dev/cu.usbmodemFA131
---------------------
Hardware ID: USB VID:PID=8087:0AB6 SER=AE6771SQ55201P7 LOCATION=250-1.3
Description: GENUINO 101

valeros, 2016-06-14:
Hi @tennismichel !
Could you please try to upload your firmware using the [latest loader](https://www.dropbox.com/s/ooru7e9bofeg5b7/tool-arduino101load-darwin.zip?dl=0)?
Just replace old one here `Users/michaelbinder/.platformio/packages/tool-arduino101load/`

tennismichel, 2016-06-14:
Hey Valeros, thanks for your help! Now I can upload the project by pressing the master reset button when the system asks for it. But without pressing i get the same output as posted above. Should I replace the whole folder for the loader or only the exec file. Both give the same results in terms of uploading.

tennismichel, 2016-06-14:
Further I still get some errors like "String is not declared in this scope" with red dots in the code, but they won't stop me from building or uploading and the program works how it's supposed to be.

ivankravets, 2016-06-15:
@tennismichel could you try the latest http://docs.platformio.org/en/latest/installation.html#development-version ?

tennismichel, 2016-06-15:
The development software didn't make a difference. Still works the same way. When I upload the project it's waiting for the device until I press the Master reset button. After that the connection works, but I can read something like "Invalid signature... " before I get the "success" message.

ivankravets, 2016-06-15:
I've just fixed it. Please re-install http://docs.platformio.org/en/latest/installation.html#development-version

It should be `2.10.2.dev1`

closed 2016-06-15 by ivankravets

tennismichel, 2016-06-15:
AMAZING, it worked! Thank you so much, Ivan!!

ivankravets, 2016-06-15:
Thank you that reported this issue. We will issue new release today.

commit 682e435d (2016-06-15): Fix firmware uploading to Arduino/Genuino 101 // Resolve #695
commit 6a08b212 (2016-06-15): Merge branch 'develop' into feature/platformio-30

* develop:
  Fix issue with ``platformio init --ide`` command for Python 2.6
  Version bump to 2.10.3
  Fix issue with appending PIO version to Build Defines
  Add test for `init --ide eclipse`
  Version bump to 2.10.2 (issues #695)
  Fix firmware uploading to Arduino/Genuino 101 // Resolve #695
  Use $PLATFORM from build environment instead from the board config
  Update Intel ARC32 Arduino framework to v1.0.6 // Issue #695
  Update Intel ARC32 Arduino framework to v1.0.6
  Add support for ST Nucleo L031K6 board to ARM mbed framework
  Revert [... cut]