platformio/platformio-core #697: cannot add a new package to a new or existing platform

boneskull, 2016-06-16:
What kind of issue is this?

- [ x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.
------------------------------------------------------------------

### Configuration

**Operating system**: MacOS X

**PlatformIO Version** (`platformio --version`): 2.10.3

### Description of problem

I apologize if I've misunderstood the docs and the source code, but it appears that the only packages that are supported are those which are prebuilt and hosted on PlatformIO's servers.

Indeed, I cannot find a reference to a download URL for any particular package in the source, because the CLI seems to hit an API endpoint for the information ("manifest").  I don't know how to circumvent this behavior or otherwise define my own download URL (without butchering several modules).

My use case:

I want to add a new board ([Oak by Digistump](http://digistump.com/oak)) with the ultimate intention of submitting a PR.  This board needs at least one custom package (a CLI tool) that is currently unsupported by PlatformIO.  It *appears* fairly straightforward to use the information from the [Arduino board index](https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json).  Unfortunately, I was almost immediately blocked due to this problem.

Due to the binding of packages to platforms, it *appears* I need to customize the `espressif` (as the Oak is ESP8266-based) and `base` platforms.  If you can, please confirm that this would be the correct strategy.

Yes, this could be a feature request for "add support for Oak".  I am perfectly capable of adding the support myself (insofar as I'm not blocking on this or future unknowns), which is why it isn't.

ivankravets, 2016-06-16:
How about to try PlatformIO 3.0? 
http://docs.platformio.org/en/feature-platformio-30/platforms/creating_platform.html

Development Platforms - https://github.com/platformio?utf8=✓&query=platform-

What is more, you can modify existing development platform or make PR to new repositories and we will merge. 
Thanks.

ivankravets, 2016-06-16:
How to install PlatformIO 3.0 https://github.com/platformio/platformio/issues/423

appcoders, 2017-04-18:
@boneskull Is this issue fixed with 3.0 ? How did you solve it. I am also stuck with the board lists but no place where to put it. Thanks.

boneskull, 2017-04-19:
@appcoder yeah I had more success with 3.0

boneskull, 2017-04-19:
this should probably be closed

closed 2017-04-19 by boneskull

appcoders, 2017-04-19:
@boneskull  that's great to hear, any hint how you solved that with 3.0 would be cool as I installed 3.0 but still struggle to add the oak board.