platformio/platformio-core #698: build_unflags does not apply to linker commands

lerneaenhydra, 2016-06-16:
What kind of issue is this?

- [ ] Bug report. If you’ve found a bug, please provide information below.

### Configuration

**Operating system**: Linux stm32dev-VirtualBox 4.4.0-24-generic #43-Ubuntu SMP Wed Jun 8 19:25:16 UTC 2016 i686 athlon i686 GNU/Linux

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.10.1

### Description of problem
The build_unflags directive does not apply to the generated linker command. For example, when using the ststm32 platform the default flags '-nostartfiles' and '-nostdlib' are **not** removed when the configuration file contains 'build_unflags= -nostdlib -nostartfiles'

#### Steps to Reproduce

1. Compile any project using a platformio.ini file containing;
[env:disco_f407vg]
platform = ststm32
framework = spl
board = disco_f407vg
build_unflags = -nostdlib -nostartfiles

### Actual Results
the following linker command is generated
`arm-none-eabi-gcc -o .pioenvs/disco_f407vg/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 **-nostartfiles -nostdlib** -Wl,-T"stm32f405x6.ld" .pioenvs/disco_f407vg/src/backend/cbuf.o .pioenvs/disco_f407vg/src/backend/dsp_wrapper.o .pioenvs/disco_f407vg/src/backend/hw/board.o .pioenvs/disco_f407vg/src/backend/hw/headphone.o .pioenvs/disco_f407vg/src/backend/hw/microphone.o .pioenvs/disco_f407vg/src/backend/hw/stm32f4_discovery.o .pioenvs/disco_f407vg/src/backend/hw/stm32f4_discovery_audio_codec.o .pioenvs/disco_f407vg/src/backend/hw/stm32f4xx_it.o .pioenvs/disco_f407vg/src/backend/hw/system_stm32f4xx.o .pioenvs/disco_f407vg/src/backend/printf.o .pioenvs/disco_f407vg/src/backend/systime/systime.o .pioenvs/disco_f407vg/src/main.o -L/home/stm32dev/.platformio/packages/ldscripts -Llib_ext -L.pioenvs/disco_f407vg -Wl,--start-group -lc -lgcc -lm -lstdc++ -lnosys -lPDMFilter_GCC -larm_cortexM4l_math .pioenvs/disco_f407vg/libFrameworkCMSISVariant.a .pioenvs/disco_f407vg/libFrameworkSPL.a -Wl,--end-group`

### Expected Results
The above linker command without the `-nostartfiles -nostdlib` flags.

ivankravets, 2016-06-16:
https://community.platformio.org/t/build-unflags-broken/392/13?u=ivankravets

> [env:frdm_k64f]
platform = freescalekinetis
framework = mbed
board = frdm_k64f
build_unflags = -fno-rtti

>I already put the env.ProcessUnFlags(env.get("BUILD_UNFLAGS")) after the framework script in platformio.py, but it´s still broken.

ivankravets, 2016-06-17:
Please re-test the latest http://docs.platformio.org/en/latest/installation.html#development-version

commit 4035b9ac (2016-06-17): Better removing unnecessary flags using ``build_unflags`` option // Resolve #698
commit 037b520b (2016-06-19): Merge branch 'develop' into feature/platformio-30

* develop:
  Use env.Flatten to manipulate with CPPDEFINES
  Update name of the OpenEnergyMonitor board // Resolve #699
  Fix PyLint warning
  Remove duplicated flags // Issue #698
  Disable dependancy info
  Revert back previous LINKFLAGS
  Better removing unnecessary flags using ``build_unflags`` option // Resolve #698
  Fix unnecessary uppercase for target includes
closed 2016-06-17 by ivankravets