platformio/platformio-core #700: avrdude bug

Trel725, 2016-06-17:
Trying to upload code to Atmega32u4 device using Platformio-build fails because of strange avrdude behavior. Problem is analogous to issue #587. 

The first attempt after opening Atom works correctly, almost any further fails. In log it's clearly seen that avrdude  reads device with mistakes (trash instead of normal CATERIN name)

```
Programmer Type : butterfly
Description     : Atmel AppNote AVR109 Boot Loader
 
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = C
Software Version = 1.0; No Hardware Version given.
Found programmer: Id = "ATERINY"; type = S
Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.
 
Programmer supports the following devices:
Device code: 0x44
 
avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions
avrdude: error: buffered memory access not supported. Maybe it isn't
avrdude: initialization failed, rc=-1
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: exit bootloader
avrdude done.  Thank you.
a butterfly/AVR109 but a AVR910 device?
Double check connections and try again, or use -F to override
this check.
```

In the same time uploading from by bare avrdude from CLI or Arduino IDE or even by pio run --target upload works without any problem.

valeros, 2016-06-17:
Hi @Trel725 !
Could you please provide more details about IDE, CLI versions, OS?

Trel725, 2016-06-17:
I am sorry for not providing information from beginning
IDE: Atom 1.8.0, from repo ppa:webupd8team/atom
OS: Ubuntu 16.04
CLI: platformio installed via pip

Also while installation I have found another bug - assignHooks in build-provider.js is not defined, so I just commented a code responsible for autoclose of serial monitor

```
if (atom.config.get('platformio-ide.autoCloseSerialMonitor')) {
         settings = settings.map(assignHooks);
        }

```

P.S. If it is more comfortable for you, I can speak Russian

ivankravets, 2016-06-17:
It looks like you have different PlatformIO CLIs. PlatformIO IDE has built-in PlatformIO CLI. What I recommend:

1. Uninstall system/global PlatformIO CLI via `pip uninstall platformio`
2. Open PlatformIO IDE and run `Menu: PlatformIO > Upgrade PlatformIO`

Trel725, 2016-06-17:
It seems to be that I don't have built-in platformio, the folder $HOME/.atom/packages/platformio-ide/penv doesn't exist.

ivankravets, 2016-06-17:
> In the same time uploading from by bare avrdude from CLI or Arduino IDE or even by pio run --target upload works without any problem.

What does it mean (` even by pio run --target upload`)? How did you run that command?

Trel725, 2016-06-17:
I run this command from standard system terminal or from Platformio IDE Terminal - it works. Obviously now it doesn't after removing platformio via pip.

ivankravets, 2016-06-17:
1. Close PlatformIO IDE
2. Please uninstall global PlatformIO CLI (`pip uninstall platformio`)
3. `rm -rf ~/.atom/packages/platformio-ide/penv`
4. Start IDE

Trel725, 2016-06-17:
In that case IDE cannot find platformio executables.  Directory ~/.atom/packages/platformio-ide/penv did not exist. Atom don't try to download platformio binaries.

ivankravets, 2016-06-17:
Do you have checked "Use Built-in PlatformIO" in the settings?

Trel725, 2016-06-17:
Yes, it was enabled all time. I also have tried to disabled it - no changes.

ivankravets, 2016-06-17:
Could you provide TeamViewer session for debug? Thanks.
me@ikravets.com

closed 2016-06-17 by Trel725