platformio/platformio-core #701: Cannot flash arduino

darkwingduck42, 2016-06-18:
- [ ] Bug report. If you’ve found a bug, please provide information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Arch Linux

**PlatformIO Version** (`platformio --version`):

### Description of problem
I get an error if try to flash the program:

/home/max/.platformio/packages/tool-avrdude/avrdude: error while loading shared libraries: libusb-0.1.so.4: cannot open shared object
file: No such file or directory

#### Steps to Reproduce

1. Create main.cpp
2. Add code
3. Try to flash

### Actual Results
Error

### Expected Results
Success

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nanoatmega328]
platform = atmelavr
framework = arduino
board = nanoatmega328
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Additional info

ivankravets, 2016-06-18:
You need to install usblib:
```
$ pacman -S libusb-compat
```

ivankravets, 2016-06-21:
Please reopen if you any questions

closed 2016-06-21 by ivankravets