platformio/platformio-core #702: Return of #48 - Problem with #include <SPI.h>

SirUli, 2016-06-19:
### Configuration

**Operating system**: Windows / Travis CI on Ubuntu

**PlatformIO Version** (`platformio --version`): 2.10.3

### Description of problem
See here for the issue: https://travis-ci.org/SirUli/Grill-Temperature-ESP8266-MAX6675/builds/138731970

Repo for the Build: https://github.com/SirUli/Grill-Temperature-ESP8266-MAX6675

If i include SPI.h in the src.src (basically the main) then it works. But if i just use the library [MAX6675](https://github.com/SirUli/MAX6675) where it is required and used it throws an Error like in the build log.

#### Steps to Reproduce & Actual Results

See Travis CI

### Expected Results

Build runs fine

### Additional info

This bug was already once in #48 where this seemed to be fixed - but it looks as if that has returned?

Thanks in advance for checking!

Best Regards,
Uli

ivankravets, 2016-06-19:
This is not a bug... It is a specific behaviour of PlatformIO 2.0. The deep library search was turned off a lot of releases ago. If you really want to enable it, then take a look at http://docs.platformio.org/en/latest/projectconf.html#lib-dfcyclic

But... I don't recommend to enable it. Please include `SPI.h` manually in one of your `src` files. I'm working on the new Library Builder that will be introduced in PlatformIO 3.0. It will be very smart with "C PREPROCESSOR" behaviour and these "issues" will be automatically fixed. 

I don't want to refactor PlatformIO 2.0 Build System because it wasn't changed significantly from the first 0.1.0 release (2014 year). PlatformIO 3.0 is really new instrument where the Build System is rewritten from the scratch.

SirUli, 2016-06-19:
Ooooh my bad - i didn't see that. Thanks for the hint, then i will just include SPI from there :)

Thanks for the quick reply!

closed 2016-06-19 by SirUli